logisim-evolution/logisim-evolution #1970: AppImage

Robin-Sch, 2024-02-26:
It would be nice if an AppImage would be published

maehne, 2024-02-28:
PRs in this direction are welcome, but are currently not of high priority for us, as Logisim-evolution is already available as Flatpak and Snap in addition to DEB and RPM packages for Linux.

piimah, 2025-12-19:
the problem is i never was able to install the deb pack bc everytime there is an error and dependencies missing. and i always try to install em but they can not be installed either cuz they have also dependencies missing and to install them i need something else and they do also not work somehow and i just come to a point of fk yu all man i just wanted a fking logisim working and have to brick and destroy my hole debian kernel everytime!

maybe you guys try to install it yourselfs to see how it is impossible. i reinstalled my debian 3 times bc of it and it still doesn't work! debian 12 and 13 same problem.

Linux 6.12.57+deb13-amd64, gnome 48, fw version 1.9.1, wayland

kailando, 2026-09-06:
I'm in favor of this, would also make for a nice portable for base Linux

maehne, 2026-09-07:
As I said above, PRs in this direction are welcome, but we lack resources to work ourselves on this issue. Note also that the Debian packaging issues #1569 and #2959 have been resolved a few days ago.