logisim-evolution/logisim-evolution #1972: All Gates - Red Output

amunrarara, 2024-02-28:
Cheers,

This is occurring on the 3 most recent releases. All projects have this issue, it's been a thing since I first downloaded, and it's negatively impacting my projects.
 
Here's some screenshots + my specs

[image]
[image]
.

davidhutchens, 2024-02-29:
Each of the output values in your picture matches the intended output for that situation.

The Switch connects the output to the input when it is on (i.e. 1) and disconnects it when it is off (i.e. 0). A disconnected output is not driven to any state and is thus floating (U), indicated by the blue color. All the switches in your picture are off (0).

If you provide inputs that have no known 0 or 1 value to a logic gate you will get an unknown result. It is not floating but an unknown logic value. That is designated in the simulator by an error (E) value displayed as red. It is a mistake to design hardware that uses such a value for further computations.

Likewise, using a U (blue) control value for a controlled buffer or inverter leaves the simulator with no way to display a meaningful output since it doesn't even know if it should drive the output or not. Thus that output is displayed as red (E) and should not be used.

So you should make sure your logic gates have well defined 0 or 1 inputs.

Read the Help for those components for more information.

maehne, 2024-02-29:
To add to @davidhutchens very good answer: If you want to continue using the Switch for controlling your inputs, you need to add Pull Resistors with Pull Direction set to Zero to ensure that the inputs of the gates have a well-defined value if the Switch disconnects the input from the Power.

closed 2024-02-29 by amunrarara