logisim-evolution/logisim-evolution #1974: wont save circuit after cancel

telephone001, 2024-03-13:
installed with logisim 3.8.0 x86 msi 
processor Intel(R) Core(TM) i5-10210U CPU @ 1.60GHz   2.11 GHz

When I open the save menu, go to recently opened and press cancel, I can't save the circuit again and I have to make it again.

davidhutchens, 2024-03-13:
I tried to reproduce this on my Mac. I was not able to do so. I wasn't sure what "go to recently opened" means. I navigated to a place with .circ files, but that didn't trigger the problem. Neither did actually opening a recently opened .circ. Could @telephone001 provide more detailed instructions to reproduce the problem? Or someone running Windows should have a look.

telephone001, 2024-03-14:
@davidhutchens 
1. open logisim and create a new project
2. at the top left, click file
3. then click save
4. for me a window that looks like this opens up.
when the save menu opens, there might be a tab called recently opened.
[image]
5.click X or close that window.
6. try to save again
for me it won't open up that save window again

telephone001, 2024-03-14:
oh wait sorry it was actually called recent items.

davidhutchens, 2024-03-15:
@telephone001: Thank you for providing more information. I tried what you suggested and the save window continues to work correctly for me. I cannot replicate the problem on the Mac.

Someone with access to Windows should try replicate it.

closed 2024-08-16 by maehne