logisim-evolution/logisim-evolution #1994: Custom wire colors / multi-wire selecting

Quantodeluz, 2024-04-24:
It would be very nice to have an option for setting custom colors for the wires in the propieties, just like the LED. I know it would be confusing if someone colors a wire like an error, but that could be solved by adding an option to highlight / mark errors. This would be very useful for organization / debugging purposes.

Also, there could be a way to select a whole wire (A wire + all the ones conected to it + all the ones conected to tthose + ...)

maehne, 2024-04-25:
You can already highlight a net by clicking with the hand tool on a wire while value propagation is active. This will also add a callout showing the nets' value. For illustration, see this contrived example:

[image]

IMHO, the capability to assign colours to wires has a too high confusion potential, as the nets' value is signalled by the colour. Therefore, this feature requires more discussion to clearly define the use cases and restrict the choices of colours to a subset, which doesn't interfere with value signalling.

PRs in this direction are welcome and will be reviewed by us, but we have to focus our limited development resources on other higher priority issues.

maehne, 2025-01-17:
@ChristoferLv: Thanks for your feedback! As stated above: "PRs in this direction are welcome and will be reviewed by us, but we have to focus our limited development resources on other higher priority issues."

jamesaharrisonuk, 2026-09-28:
I find it very difficult to follow traces when they are the same colour. Different colours would make it easier to follow a trace. I don't know enough about the code to commit a MR change.

V-Zemlyakov, 2026-09-28:
Unfortunately, implementing custom colors for individual wires is technically unfeasible right now due to the core architecture. As I explained in a similar issue regarding wire attributes (https://github.com/logisim-evolution/logisim-evolution/issues/2955#issuecomment-5555695181), wires in Logisim work differently than regular components. 

They are essentially just basic lines (`Wire.java`) and do not store attributes internally. Any additional visual property has to be managed separately in `CircuitWires.java`. Adding custom color tracking would require massive changes to the core engine logic (`Clipboard`, `SelectionBase`, `CircuitMutation`), bringing a very high risk of breaking fundamental features like undo/redo and copy/paste.

However, as a compromise to help with circuit readability, I am planning to add the ability to apply color highlighting (a custom colored background) to **Tunnels** and **Pins**. While it's not exactly coloring the wires themselves, it should help you visually group and track specific signals across complex schematics without touching the fragile wire engine.

Would this be a helpful alternative for your workflow?

maehne, 2026-09-28:
Instead of making the wire color an attribute, how about implementing a functionality similar to [net probes in Xschem](https://xschem.sourceforge.io/stefan/xschem_man/net_probes.html)? In that schematic entry tool, one can highlight one or more nets during run time. Each net receives automatically a distinct colour. The nice thing is that the highlighting works across the whole circuit hierarchy. It is very helpful to visually inspect the correct wiring of a circuit. This run time functionality is helpful while creating/debugging circuits as well as demonstrating them to an audience. IMHO, saving the highlighting with the circuit is less useful and has even the potential of confusing people with additional clutter -- especially when transferring the circuit to someone else or reusing an old circuit after some time.

@V-Zemlyakov: What do you think? Would this be more easily implementable with our current architecture? Of course, including Pins and Tunnels in the net highlighting would make perfectly sense.