logisim-evolution/logisim-evolution #2002: No X11 DISPLAY variable was set - Wayland

VehementHam, 2024-05-13:
I get this error when running on Wayland:
```
May 12, 2024 7:42:59 PM java.util.prefs.FileSystemPreferences$1 run
INFO: Created user preferences directory.
Exception in thread "main" java.awt.HeadlessException: 
No X11 DISPLAY variable was set,
or no headful library support was found,
but this program performed an operation which requires it,

	at java.desktop/java.awt.GraphicsEnvironment.checkHeadless(GraphicsEnvironment.java:166)
	at java.desktop/java.awt.Window.<init>(Window.java:553)
	at java.desktop/java.awt.Frame.<init>(Frame.java:428)
	at java.desktop/java.awt.Frame.<init>(Frame.java:393)
	at java.desktop/javax.swing.SwingUtilities$SharedOwnerFrame.<init>(SwingUtilities.java:1925)
	at java.desktop/javax.swing.SwingUtilities.getSharedOwnerFrame(SwingUtilities.java:2001)
	at java.desktop/javax.swing.JOptionPane.getRootFrame(JOptionPane.java:1696)
	at java.desktop/javax.swing.JOptionPane.showOptionDialog(JOptionPane.java:875)
	at java.desktop/javax.swing.JOptionPane.showMessageDialog(JOptionPane.java:677)
	at java.desktop/javax.swing.JOptionPane.showMessageDialog(JOptionPane.java:648)
	at java.desktop/javax.swing.JOptionPane.showMessageDialog(JOptionPane.java:619)
	at com.cburch.logisim.gui.generic.OptionPane.showMessageDialog(OptionPane.java:45)
	at com.cburch.logisim.Main.main(Main.java:65)
```

I am running the `.jar` file on Hyprland Alpine Linux.

maehne, 2024-05-15:
You likely don't have a full JDK installed on your machine, but a reduced "headless" variant. See, e.g., [this discussion](https://discussion.fedoraproject.org/t/cannot-run-java-jar-file-in-wayland-only-fedora-40/111814) on the Fedora forum. The binary packages, which we provide, e.g., as a [nightly build](https://github.com/logisim-evolution/logisim-evolution/actions/runs/8993940035) contain a bundled JDK. If you build from source or use the shadow JAR, you have to make sure that you have a full OpenJDK 21 installed, e.g., from the [Eclipse Temurin project](https://adoptium.net/de/temurin/releases/).

VehementHam, 2024-05-24:
I think that was it.

closed 2024-05-24 by VehementHam