logisim-evolution/logisim-evolution #2005: Simulation halted by internal error when generating timing diagram

arkis-s, 2024-05-15:
I'm trying to create a timing diagram of a CPU whilst its executing a program, however it fails with the following error:
```
java.lang.ArrayIndexOutOfBoundsException: Index 2 out of bounds for length 2
        at com.cburch.logisim.gui.log.Signal.extend(Signal.java:95)
        at com.cburch.logisim.gui.log.Model.extendWithNewValues(Model.java:570)
        at com.cburch.logisim.gui.log.Model.updateSignalsClockMode(Model.java:671)
        at com.cburch.logisim.gui.log.Model.propagationCompleted(Model.java:598)
        at com.cburch.logisim.gui.log.LogFrame$MyListener.propagationCompleted(LogFrame.java:86)
        at com.cburch.logisim.circuit.Simulator.firePropagationCompleted(Simulator.java:482)
        at com.cburch.logisim.circuit.Simulator$SimThread.loop(Simulator.java:373)
        at com.cburch.logisim.circuit.Simulator$SimThread.run(Simulator.java:383)
```

And it also produces a pop-up window that tells me to save my work and reload Logisim. I'm not able to restart the simulation using ``CTRL+E``
[image]

I am using v3.8.0 installed using the .deb package.

**To reproduce:**
1. Download ``BABY.circ`` and ``TuringLongDivision.logisimRAMImage`` from https://gitlab.com/charles.fox/comparch/-/tree/main/chapter07
2. Open ``BABY.circ`` and load the ``TuringLongDivision.logisimRAMImage`` into the RAM block
3. Change the frequency: ``Simulate`` -> ``Auto-Tick Frequency`` -> ``16.0 Hz``
4. Open the timing diagram utility: ``Simulate`` -> ``Timing diagram``
    1. Enable ``Store unlimited history``
    2. Set timescale to 5.0us
    3. Do not add/remove any signals
 5. Click into the ``Timing diagram`` tab in the utility
 6. Start the simulation: ``Simulation`` -> ``Auto-Tick Enabled``
 7. Wait for it to crash, happens when the chronogram hits 2ms

[image]

I suspect that turning on ``Store unlimited history`` might be the cause of the issue, as the simulation runs as intended with it disabled, but I was hoping to export the waveforms generated during the entire duration of the program execution.

maehne, 2024-05-16:
Thanks for the report! Could you please first test with [the latest nightly build](https://github.com/logisim-evolution/logisim-evolution/actions/runs/9104282361) whether the issue persists? There have been quite some bug fixes since the last official release.

maehne, 2024-05-16:
@arkis-s: Thanks for your testing with the latest nightly! I am not sure how fast we'll be able to fix the root cause, as the chronogram code was not developed by the current main developers. Maybe, @kevinawalsh already bumped on this bug in his Holy Cross fork of Logisim-evolution, as issue https://github.com/kevinawalsh/logisim-evolution/issues/7 sounds similar.

arkis-s, 2024-05-16:
Fair enough, thanks for letting me know :) In the meantime I'll just export the HDL and use it in some other simulator.

closed 2025-05-29 by BFH-ktt1