logisim-evolution/logisim-evolution #2006: Debian build and Linux release value

davidhutchens, 2024-05-17:
As I described in #1997, Java's jpackage utility changed the default filename for the Debian package to drop the release number as the default. The Java issue describing what they did and why is [here](https://bugs.openjdk.org/browse/JDK-8276084). They also added an option to jpackage, --linux-app-release, to have it include the release number optionally. To restore the original format, we would need to use that option and change Nightly to look for the correct filename.

I took the quick path to get our Nightly Build working again by changing Nightly to look for a name without the release field. If the Java issue correctly states the situation, we can leave it as it is now.

I am opening this issue so we can discuss restoring the release field if anyone feels that we should. It also documents why it was changed and states how to return to the previous behavior should that be desired or necessary in the future.

Comments? @maehne? @BFH-ktt1?

maehne, 2024-05-17:
I would leave the DEB package name to the recommended default without the release-number. The latter is anyway only necessary if the packaging gets improved for the same software release version so that `apt` knows that an update of the installed package is due.

maehne, 2024-05-20:
@BFH-ktt1: Do you agree to drop the release number for DEB packages? Then, I think this issue can be closed as resolved.

BFH-ktt1, 2024-05-28:
I agree with the drop.

davidhutchens, 2024-05-28:
Closing as we agree no further action is needed.