logisim-evolution/logisim-evolution #2011: Splitter as Input / Output sub circuit port

ShuffleWire, 2024-06-01:
Hello Logisim Folks !

NOTE : This post don't request to add input/output pins ! It have been already discussed elsewhere (I know it required simulation engine rework), and it's not the feature I'm looking for.

I'm designing a communication node, that take 2 pairs of inputs and output each. To make things clear, lets assume that each input and output is 1 bit. _(In my real work, there are 4 pair of input/output, and each input/output is 10bits, but whatever)_

Those node should be chained together to pass information from A to B and from B to A. Lets draw it in Logisim : (the communication node is dummy, just a passtrought, in my works it's doing useful stuff !)

[image]
[image]

Now lets build a "network" by linking them together.

[image]

It work just fine, and info it passed through each node, IN BOTH DIRECTION using DIFFERENT PINS.

But as you could imagine, I do want more complex network, with node that don't align (remember, in my work, there are, for each node, INPUT_C, OUTPUT_C, INPUT_D, OUTPUT_D, to the top and bottom, so network will span in the 2D plan quickly)

I could get something like that : 
[image]

It's not very clean to have to manually manage those pair of wire.

For the moment my solution is to use Splitter at each INPUT/OUTPUT, to have only one wire to manage, like so : 
[image]

It's work just fine, because Splitter could transmit data in BOTH DIRECTION at the same time.
However, it's not very convenient to have to add splitter manually. I wish I could push them INTO the node (or at least using a intermediate circuit), like so : 
[image]
[image]

But this, obviously, DON'T work... Mainly because the pin in the new node (those who are 2 bits wide) are INPUT for one bit, but OUTPUT for the other.
Note that each bit, is by itself clearly a INPUT or an OUTPUT, but the combination of those two confuse Logisim.

So, could we find a solution ? (Do I missed something ?).

I give you the full file presented in the screenshot (I need to zip it...)
[logisim_splitter.circ.zip](https://github.com/user-attachments/files/15522724/logisim_splitter.circ.zip)

Thank you so much for reading me up to that point,

Cheers.

maehne, 2024-06-05:
Thanks for the detailed feature request! I am not knowledgeable enough about the implementation to judge whether this can be implemented with reasonable effort within the constraints of the current simulation engine. PRs in this direction are certainly welcome!

closed 2025-06-01 by BFH-ktt1

ShuffleWire, 2025-06-02:
Even without knowing the code, I would argue that my feature is different and simpler than bidirectional pin : for bidirectional pin, the pin should be able to work "dynamically" like input or output.
However, my feature is only about grouping different kind, but fixed, type of pins. Maybe create a type on input/output pin specifically designed to handle that case. It could even enforce statically the kind of input/output combination (like 2 input and 1 output in a 3 way splitter), like we currently do on input/output pin (for instance, I can't connect a wire containing 4 signals to a pin expecting 6 wire, as the bit width will not match)

PS: What if I could grant some money for someone willing to take on it ?

maehne, 2025-06-02:
@ShuffleWire: I am afraid that @BFH-ktt1 is right, implementing your feature request would not only have a drastic impact on the simulation engine, but also the HDL netlisting to VHDL and Verilog. Also in these languages, each port get assigned a direction upon declaration. Even when using `record`s in VHDL to model complex interfaces, typically two separate `record`s are employed to consider the two directions of signal flow -- as you currently do in your Logisim circuits. Otherwise, one would need to use `inout` for the port direction and the compiler cannot any more check the port direction for the individual signals of the interface. A good solution would be [VHDL 2019 interfaces](https://www.sigasi.com/tech/what-is-new-in-vhdl-2019-part2/), but many tools aren't yet supporting this recent language version.

As we need to prioritize our limited development resources, we cannot work on this feature. If someone steps up to work on this feature and bidirectional ports, we will be more than happy to review his PR(s). However, it is a considerable task, which does touch many parts of Logisim-evolution and its library of component models. I hope that you can understand our position!

ShuffleWire, 2025-06-02:
Thank @maehne for your detailed explanation. I do completely get your position, no worries ! I will continue to work on my design using Logisim anyway :)  
Cheers !