logisim-evolution/logisim-evolution #2013: "Unconnected" State don't get propagated to Subcircuit

ShuffleWire, 2024-06-02:
Hello,

I'm working with subcircuit that have some "optional" input.
On my first try, I forgot to add some king of pull down in case it's not connected. But debugging that lead me to an strange state.

I link you the full file [test.circ.zip](https://github.com/user-attachments/files/15525363/test.circ.zip)

Basically I left Unconnected the input pin of my subcircuit, but during the simulation, it's get converted to a 0.

[image]

Even if it could be seen as helpful, it did hide my issue in my design for some time, and I don't think it's the expected behavior anyway.

Did I miss something ?

Cheers

maehne, 2024-06-05:
Thanks for the bug report! Indeed, the `U` should get properly propagated. PRs to fix this issue are welcome!

davidhutchens, 2025-03-23:
The problem here is that the input pin has the Three-State? attribute set to No. If you set it to Yes, it passes the U value to the output. This setting is not easy to understand since it allows the U input from the superstate and even displays it in the substate.

I noticed while working with his fork that Keven Walsh redesigned the attributes of the Pins to make it easier for the user to tell what the settings mean. Perhaps we should consider that. There is a long comment at the end of Pin.java in his fork explaining how it used to work and what he changed it to.

maehne, 2025-03-26:
Thanks @davidhutchens for your analysis of the issue! I agree that backporting the redesigned Pin attributes from @kevinawalsh's Holycross fork would be the good step forward. His comments clarify well the motivation behind the done changes. Ideally, the new behaviour should be also reflected in the online help. Would you like to assign yourself to this issue?

In case you should do a PR with backported commits, please document them as well in the tracking issue #231.

davidhutchens, 2025-05-16:
I have ported the changes from the Holycross fork. I haven't yet looked at the online help.

@maehne and @BFH-ktt1: I have a question about reading files with the old pin attributes, specifically the one with no attributes changed, like the one pictured and discussed in this issue.

The way @kevinawalsh implemented the Pin in his fork, the behavior of the default input pin is changed to pass through the U, while the old default pin pulled U to 0. I consider this a fix of a long-standing bug since the pin should not be pulling to 0 when the pull attribute is not set to pull to 0. However, since that is a change in behavior, it will break any existing circuit that incorrectly depended on the old behavior.

Kevin's approach is to let the user fix any pins that should have been pull-down but were not. There is no way to figure out while reading the file which pins should have had pull-down selected, so any attempt at automatically fixing the pins would apply the pull-down attribute to very large numbers of pins that do not need it. Kevin's approach seems reasonable to me. I would, of course, note the issue in CHANGES.md.

Is it acceptable to leave this non-backward-compatible change in this PR, or do I need to figure out a way to maintain full backward compatibility?

maehne, 2025-05-27:
Thanks for your effort @davidhutchens and sorry for the delayed answer to your question! I completely agree with @kevinawalsh and your reasoning! It is sufficient to fix the `U` propagation and document the change in behaviour in the `CHANGES.md` file mentioning the setting to fix the issue in the unlikely case that someone relies in his circuit on such behaviour.

closed 2025-06-01 by BFH-ktt1