logisim-evolution/logisim-evolution #2016: the .jar file is not working.

hix1000, 2024-06-12:
I just get an error message that says "A java exception has occurred"

davidhutchens, 2024-06-12:
The most likely problem is that you need to upgrade Java. You need at least Java 21 for our daily build and at least Java 16 for the latest release (3.8.0).

The next thing to try is download the .jar file again. Perhaps yours was corrupted. Or try the package build for your platform since it includes the correct Java version.

If none of those things work, please describe your system, Java version, and which package(s) you tried.

davidhutchens, 2024-06-19:
Since we have no further information, I assume this issue is resolved. If not, please reopen it with appropriate detail.

closed 2024-06-19 by davidhutchens