logisim-evolution/logisim-evolution #2021: Impromer byte-enable behaviour on RAM module

suehtemorp, 2024-06-24:
Hi! This is the first issue I open on here, and I'm pretty new to opening issues in general. Please be patient with me on issue-etiquette / decorum :)

I've been using the RAM module for a college assignment and I've noticed the byte-enable ports don't work as expected on a 16M x 64 spec (24-bit addresses, 64-bit words).

**Desired behavior:**
- The RAM module allows for a byte-enable's selection to be toggled on the properties tab.
- If the option is toggled on, every write attempt should only allow the updating of the bytes that have their corresponding byte-enable port fed with a true logical value. This applies for the currently-selected word (via the address port)

**Observed behavior**, (given that the byte-enabled's selection is toggled on):
- The byte-enable ports 0-2 (labeled M4 to M6) work as expected.
- If only the byte-enable port 4 (M7) is allowed to be fed a true value, all but those bytes associated with the byte-enabled ports 0-3 are updated.
- If only the byte-enable ports 5-7 (M8-M10) are allowed to be fed true values, they behave as if they were the ports 0-2.
- If only the byte-enable port 8 (M11) is allowed to be fed a true value, it behaves like the port 4 behaves currently.

**Suspected cause:**
Looking at main, specifiically at 30ad0b062c5aeb9cd05bee16be4fd615e51cd61d, (here's a [permalink](https://github.com/logisim-evolution/logisim-evolution/blob/84be7a6aafbc6f0198cc0d479777391f8e3c665b/src/main/java/com/cburch/logisim/std/memory/Ram.java)), I believe the error comes from a missed literal-cast.

It ocurrs here:
https://github.com/logisim-evolution/logisim-evolution/blob/84be7a6aafbc6f0198cc0d479777391f8e3c665b/src/main/java/com/cburch/logisim/std/memory/Ram.java#L331-L336

Instead of `0xFF` I believe the mask should use  `0xFFL` to enforce a cast. One can check the difference in behaviour with a simple Java compiler.

maehne, 2024-06-25:
Thanks for reporting the issue and investigating the cause! Your suggested fix looks good. Would you mind opening a PR?

suehtemorp, 2024-06-26:
Sure! Opened on #2023.
(I'm sorta new to doing PRs so let me know if I missed something from the contribution guidelines.)

maehne, 2024-06-26:
Thanks for opening the PR!

suehtemorp, 2024-06-28:
No problem! :)

BFH-ktt1, 2024-07-05:
Closing as PR got merged.

closed 2024-07-05 by BFH-ktt1