logisim-evolution/logisim-evolution #2028: How do I download the latest changes?

SLicudis, 2024-07-09:
I've got the "newest version" of L.E. (3.8.0)
I was looking for the 74181 in the TTLs folder but I couldn't find it. The most recent changelog says that 74181 was added:
[image]
Could you please help me to get the newest version of logisim?

SLicudis, 2024-07-09:
(Btw, I'm on windows 11 x86-64)

davidhutchens, 2024-07-10:
The latest build is at [https://github.com/logisim-evolution/logisim-evolution/actions/workflows/nightly.yml](https://github.com/logisim-evolution/logisim-evolution/actions/workflows/nightly.yml). To find an actual build, you need to look down the list to find one that took a several minutes to run. If it took less than a minute, there were no changes that day and the build was skipped.

For example, right now, you can get a recent nightly build at build number 1175. For x86-64 Windows, look for the amd64.msi installer.

davidhutchens, 2024-07-13:
I am closing this as answered.

closed 2024-07-13 by davidhutchens