logisim-evolution/logisim-evolution #2030: Ideas for the 3.9.0 update

SLicudis, 2024-07-14:
Here are my ideas for 3.9.0:

-Adding a Telnet I/O device: Digital has a telnet device. It's extremely useful and easier to use than the Keyboard and TTY

-"Compiling" circuits to simulate: This feature will probably be hard to implement. It consists on making a simulation mode in which everything runs faster but with the cost of not being able to modify things during the simulation

-Don't make it mandatory to use the Ctrl key for actions: I saw that in the in-dev version of 3.9.0, you have to press Ctrl + F9 to step the clock, and I can't make key shortcuts without using Ctrl.

Cosmic0921, 2024-07-14:
wait there's a v3.9.0 in dev build? I didn't know about that

maehne, 2024-07-26:
@Cosmic0921: You can always check out the latest [nightly build](https://github.com/logisim-evolution/logisim-evolution?tab=readme-ov-file#nightly-builds) to test the current development state in Git.

maehne, 2024-07-26:
@SLicudis: Thanks for suggesting features for the next Logisim-evolution release. However, our resources to maintain Logisim-evolution are limited. We are happy to review/integrate pull requests in this direction.

> -Adding a Telnet I/O device: Digital has a telnet device. It's extremely useful and easier to use than the Keyboard and TTY

See my reply to your [question](https://github.com/logisim-evolution/logisim-evolution/discussions/2029). 

> -"Compiling" circuits to simulate: This feature will probably be hard to implement. It consists on making a simulation mode in which everything runs faster but with the cost of not being able to modify things during the simulation. "Digital" manages to run things at 13 KHz, while Logisim Evolution only runs things at 150-300 Hz

With some effort, you can already do that for all components, which are synthesisable, as the whole design can be netlisted to VHDL, and then you could instantiate it into a test bench for simulation in a VHDL simulator. 

Indeed, there's potential to improve the simulation performance, see issue #786 and PR #1805. 

> - Don't make it mandatory to use the Ctrl key for actions: I saw that in the in-dev version of 3.9.0, you have to press Ctrl + F9 to step the clock, and I can't make key shortcuts without using Ctrl.

If you check the Hotkey Settings tab, you will note that it is a conscious choice to require Ctrl for menu shortcuts. I agree that not requiring Ctrl for shortcuts related to controlling the clock ticks/simulation might 
be more convenient.

Our development resources are currently limited. We strive for a new release before the autumn term starts, but it will probably mostly contain the features/fixes that are already on the [main](https://github.com/logisim-evolution/logisim-evolution) development branch and fixes/enhancement PRs we will receive up till the date where we have to make the cut to prepare the release.

RoyRoki, 2024-08-29:
hey i am new here

maehne, 2025-01-03:
Closing this issue due to inactivity and release of Logisim-evolution 3.9.0. The Telnet component was implemented in PR #2104. Regarding the keyboard shortcuts, you may consider to create a focused enhancement proposal as an issue.

closed 2025-01-03 by maehne