logisim-evolution/logisim-evolution #2033: ShiftRegister strange behaviour when Appearence attribute is set to "Classic Logisim"

iagoggurgel, 2024-07-23:
From what I've noticed, the Shift Register when set to "Classic Logisim" appearence display some error-like and strange behaviour in the following:

1. When you set the component to Classic Logisim, you can see the input and output ports are being set in the wrong way.
[image]
2. Whenever you change the "Number of Stages" attribute to some higher value and come back to default (8 stages), the error seems to go away.
3. Whenever you move the component around, It's ports are changing from Input to Output and vice-versa, and bugging out in general.

Appreciate the attention!

maehne, 2024-07-26:
I am not able to reproduce your issue that "the input and output ports are being set in the wrong way". Once, I connect wires to the shift registers ports, the data inputs are on top and the data outputs on the bottom. However, be aware that there is additionally a "load" control input on the top left side and a clear input on the bottom left side.

[image]

What I have noticed and differs from the Logisim-evolution appearance is the unassigned output on the bottom right side. One would expect it to hold the serial output value, which is actually on the right side. The Logisim-evolution appearance doesn't have this seemingly useless output:

[image]

Unfortunately, the documentation of the component in the is not helpful either. The description in the Logisim-evolution library reference is unfortunately inconsistent by using a picture of the Classic Logisim appearance and describing the ports based on the Logisim-evolution appearance. @mbaillif: Could you maybe have a look at this?

@BFH-ktt1 and @davidhutchens: What's your opinion on the additional unassigned output port?

maehne, 2024-07-26:
To reduce doubts how to use this component, all ports should have an associated tooltip. Currently, only the control inputs have tooltips, but the data inputs/outputs don't have any tooltip.

davidhutchens, 2024-07-27:
I checked the behavior of original Logisim with the shift register. It always provides the same value on the least significant bit of the output and the serial output pin. While it seems unnecessary to have both, I would suggest copying the output to both pins to be compatible with original Logisim. I am assuming that would not be hard to do.

maehne, 2024-07-31:
Thanks for your analysis @davidhutchens! I agree with you.