logisim-evolution/logisim-evolution #2036: RAM does not extern loaded content if placed in a subcircuit.

phbernardi, 2024-07-26:
If values are loaded manually to the RAM, this contend cannot be loaded/seen if the RAM is in a subcircuit.

For example, I made a circuit with a RAM initialized with data 0x5f5f5f20 on address 0x000

[image]

When the output is read on main circuit, the output is 0x00000000 instead 0x5f5f5f20:

[image]

This happens for any address, not just the address 0x000.

Meaning, if the RAM is placed in a subcircuit, there is no way to use an loaded initialization on RAM. Tested in version v3.8.0

maehne, 2024-07-26:
Thanks for the report! Could you please try with the [latest nightly build of Logisim-evolution](https://github.com/logisim-evolution/logisim-evolution/actions/runs/10068136448)? If the error still occurs, please attach here the `.circ` file of your test design as a ZIP archive.

phbernardi, 2024-07-27:
Confirmed, this behavior also happens with nigtly build.

[RAM_test.zip](https://github.com/user-attachments/files/16400274/RAM_test.zip)

davidhutchens, 2024-07-27:
I am not able to reproduce your pictures with your circuit. Except that I note that the output is synced with the clock. If I have not caused a low-high transition in the clock, no value is passed out of the RAM. So if you open your circuit, tick the clock to get the 0 output in main, then put the values in RAM and trigger the rd line in main, nothing new will be output until the clock is ticked. On the next tick, the desired value from address 0 appears in main.

If you are ticking the clock and seeing no value change, could you please describe exactly what steps you took after loading the circuit to demonstrate the problem?

Finally, let me note that when examining subcircuits during simulation, you should be sure to have Simulate selected rather than Design and make sure you are examining the branches of the tree from main (expanded with the >) and not the separate version of the subcircuit, which (if it exists) was treating itself as the top level of simulation.

phbernardi, 2024-07-27:
> Finally, let me note that when examining subcircuits during simulation, you should be sure to have Simulate selected rather than Design and make sure you are examining the branches of the tree from [...]

Ok, this seems to be the point, I was really using the "design" tab to perform the simulation. When using the tab "simulation" instead design I have two different memories to insert contend - one for subcircuit and one for the main circuit. Thank you for pointing this out, you can close this ticket.

davidhutchens, 2024-07-27:
Closing as answered.

closed 2024-07-27 by davidhutchens