logisim-evolution/logisim-evolution #2047: Wrong commit tagged for 3.9.0 release

jonicho, 2024-08-15:
For the new 3.9.0 release the commit 83c38fdb2f3002255e0bdd06b80035bf270b20a4 was tagged as v3.9.0, instead of the commit 2d5f2a84775a5c4ca6d431fa7b6415fca020902b, where the version was actually changed from 3.9.0-dev to 3.9.0. This results in the code that is tagged as the 3.9.0 release still saying the version is 3.9.0-dev

MarcinOrlowski, 2024-08-24:
@BFH-ktt1 if that's valid report, here are steps to fix that:

```
# Step 1: Delete the incorrect tag locally
git tag -d v3.9.0

# Step 2: Recreate the tag on the correct commit
git tag v3.9.0 2d5f2a8

# Step 3: Force push the corrected tag to the remote repository
git push --force origin v3.9.0
```

BFH-ktt1, 2025-05-29:
@MarcinOrlowski : Thanks! Fixed, closing this issue.

closed 2025-05-29 by BFH-ktt1