logisim-evolution/logisim-evolution #2048: Remove unused branches master and rc3.8.0

maehne, 2024-08-15:
We still have the unused branches `master` and `rc3.8.0`. I would like to suggest deleting [them](https://github.com/logisim-evolution/logisim-evolution/branches/all).

davidhutchens, 2024-08-15:
I agree that these are no longer needed. I was about to suggest deleting master since its existence leads to some confusion.

But having a branch for the latest release is useful. We should also create one for 3.9.

maehne, 2024-08-16:
IMHO, I don’t see a need for a branch for the current version, as there is already a tag `v3.9.0` in place for reference. It would only become necessary in case we need to apply hot fixes for some urgent issues. If `main` shouldn’t be in a state for release when that becomes necessary, then we can always create it quickly.

davidhutchens, 2024-09-10:
I agree with @maehne that keeping a version branch is not necessary.

rc3.8.0 still exists and should be removed.

maehne, 2025-01-03:
I have deleted the stale branch `rc3.8.0`, as we don't support older releases due to resource constraints.

closed 2025-01-03 by maehne