logisim-evolution/logisim-evolution #2052: Nested subcircuit custom appearance doesn't hold

CarrotHandsome, 2024-08-25:
I am trying to use subcircuit A that itself uses subcircuit B. B uses a custom appearance which shows correctly when I open A.circ. However when I place A into my main circuit and inspect it, B uses the standard appearance and ends up breaking the circuit.

maehne, 2024-08-25:
Thanks for the bug report! Did you make sure to select the right appearance when you instantiated the subcircuit? Please attach in a ZIP archive a circuit example exposing the problem and maybe some illustrative screenshots.

CarrotHandsome, 2024-08-25:
Thank you and yes in regards to selecting appearance. Now that I'm looking at it again actually, I'm finding that the appearance selection of a subcircuit is being reset to default when I close and reopen the circuit (after changing appearance to custom and then saving). I guess this must be related? I'll attach ALU.circ (subcircuit A in the explanation) and full_adder.circ (subcircuit B).

[Here is a video demonstration of the appearance not saving](https://www.youtube.com/watch?v=VyojwOKAisI)

[logisim subcircuit appearance not saving.zip](https://github.com/user-attachments/files/16741955/logisim.subcircuit.appearance.not.saving.zip)

maehne, 2024-08-26:
Thanks for the video and the reproducer! I am able to reproduce the issue on my side. It seems that the appearance of the instantiated sub-circuits gets always loaded from the library even if you overruled it when you instantiated the sub-circuits. If you select in `full_adder.circ` the `full_adder` sub-circuit in the library pane, choose Appearance "Custom", and then save  `full_adder.circ`, you will see that after opening `ALU.circ`, the instances then will have the right appearance. This issue is at least to some extent related to issue #1780. IMHO, the custom appearance requires some attention to resolve these [remaining usability problems](https://github.com/logisim-evolution/logisim-evolution/issues?q=is%3Aissue+custom+appearance+is%3Aopen). Though, we need to pay attention to not cause regressions with the [closed issues](https://github.com/logisim-evolution/logisim-evolution/issues?q=is%3Aissue+custom+appearance+is%3Aclosed) around this topic.

Any help/PRs are welcome! @BFH-ktt1 and @davidhutchens: Could you maybe have a closer look?

CarrotHandsome, 2024-08-27:
I love a simple solution I should have already tried!

PR #2253 (2025-06-01): Circuit instances cannot have a appearance attribute, as this is set by the circuit itself (static attribute)
fixes #2052
closed 2025-06-01 by BFH-ktt1