logisim-evolution/logisim-evolution #2058: ERROR com.cburch.logisim.util.SocketClient

RoyRoki, 2024-09-06:
I use Ubuntu24

`roki@rocket:~$ logisim-evolution
Picked up _JAVA_OPTIONS: -Duser.home=/home/roki/snap/logisim-evolution/4
Gtk-Message: 01:35:28.557: Failed to load module "canberra-gtk-module"
[AWT-EventQueue-0] ERROR com.cburch.logisim.util.SocketClient - Cannot create server socket
java.net.SocketException: Permission denied
	at java.base/java.net.ServerSocket.createImpl(ServerSocket.java:320)
	at java.base/java.net.ServerSocket.getImpl(ServerSocket.java:294)
	at java.base/java.net.ServerSocket.bind(ServerSocket.java:388)
	at java.base/java.net.ServerSocket.<init>(ServerSocket.java:274)
	at java.base/java.net.ServerSocket.<init>(ServerSocket.java:167)
	at com.cburch.logisim.util.SocketClient.<init>(SocketClient.java:45)
	at com.cburch.logisim.vhdl.sim.VhdlSimulatorTop.<init>(VhdlSimulatorTop.java:47)
	at com.cburch.logisim.proj.Project.getVhdlSimulator(Project.java:370)
	at com.cburch.logisim.vhdl.gui.VhdlSimulatorConsole$VhdlSimState.stateChanged(VhdlSimulatorConsole.java:62)
	at com.cburch.logisim.vhdl.gui.VhdlSimulatorConsole.<init>(VhdlSimulatorConsole.java:99)
	at com.cburch.logisim.gui.main.Frame.<init>(Frame.java:190)
	at com.cburch.logisim.proj.ProjectActions.createFrame(ProjectActions.java:141)
	at com.cburch.logisim.proj.ProjectActions$CreateFrame.run(ProjectActions.java:64)
	at java.desktop/java.awt.event.InvocationEvent.dispatch(InvocationEvent.java:318)
	at java.desktop/java.awt.EventQueue.dispatchEventImpl(EventQueue.java:771)
	at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:722)
	at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:716)
	at java.base/java.security.AccessController.doPrivileged(AccessController.java:399)
	at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:86)
	at java.desktop/java.awt.EventQueue.dispatchEvent(EventQueue.java:741)
	at java.desktop/java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:203)
	at java.desktop/java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:124)
	at java.desktop/java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:113)
	at java.desktop/java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:109)
	at java.desktop/java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.desktop/java.awt.EventDispatchThread.run(EventDispatchThread.java:90)

`

maehne, 2024-09-06:
How did you install Logisim-evolution? Did you use the DEB package for Ubuntu?

RoyRoki, 2024-09-07:
I Download the latest version of the Logisim Evolution .jar file from the official GitHub repository:

maehne, 2024-09-10:
Have you installed a recent enough and *complete* JDK, e.g., `openjdk-21-jdk`? The JAR offered for download is built against OpenJDK 21.

RoyRoki, 2024-09-11:
I have Java Archive (JAR) version 21.0.4 installed on my system

maehne, 2024-09-15:
@BFH-ktt1: Could you maybe check our JAR on Ubuntu 24.04 to check whether we can reproduce the issue?

BFH-ktt1, 2025-05-27:
Closing issue as I cannot reproduce it on my Ubuntu 24.04 LTS installation. Please reopen if the problem persists.

closed 2025-05-27 by BFH-ktt1