logisim-evolution/logisim-evolution #2059: TTY has text hanging out of it's box.

AbbysalSoul, 2024-09-07:
So, it's so weird.
I've been messing around with components out of boredom and found this.
[image]

maehne, 2024-09-08:
Please provide more details about the used processor platform, OS version, JDK version, fonts, Logisim-evolution version and its package form. A reproducer design would be also helpful to facilitate reproducing the issue.

AbbysalSoulBackup, 2024-09-15:
okah yippee:1!1!1!1

MY7H404, 2024-10-07:
I tried this on the latest Logisim-Evolution Java version on Linux, and the same thing happened to me

## How to Reproduce

It seems that when you zoom in and out this happens
- I set zoom to x1.2 and the text is normal 
[image]

- When i zoom in to x1.3, the text goes outside the box
[image]

maehne, 2024-10-23:
I can confirm your findings. Depending on the current zoom factor of your circuit sheet, the output may overlap the boundaries of the TTY component. The TTY component needs to allocate more horizontal space per column. However, the actual space needed per character depends on the actual font and its size used in a particular zoom factor.

A PR to fix this visual glitch is welcome!

shri-acha, 2025-02-05:
Hi, I've created PR, please check it out.

closed 2025-05-27 by BFH-ktt1