logisim-evolution/logisim-evolution #2060: Version 3.9.0 can not show some font

zengfan1992, 2024-09-07:
Version 3.9.0 can not show some font, but version 3.8.0 doesn't have this problem. useing same version jdk.
[image]

maehne, 2024-09-08:
Could you please also try with our native installers for your platform? Please, also provide more details about your platform to enable us reproducing the issue locally (processor architecture, OS version, set locale (i.e., language environment), used JDK version, exact Logisim-evolution version, its package form and origin, etc.). Please, provide also details about the fonts you are using. As we are not familiar with Asian languages/scripts, it will be hard for us to debug/solve the issue. Help and PRs fixing the issue are welcome.

closed 2024-09-25 by zengfan1992

maehne, 2024-09-25:
@zengfan1992: Were you able to resolve the issue on your side? If yes, it would be good, if you could briefly describe the cause of the issue and how you resolved it. This would help others stumbling on the same issue.