logisim-evolution/logisim-evolution #2061: Deb package install command fails specifying dependency errors

Animus-Surge, 2024-09-11:
Installing the deb package fails when using the `dpkg -i` command, specifying dependency issues. Attempting to install the latest release as of writing this issue, v3.9.0

### System info

Debian 12.6 x86_64; Java 21.0.3

### `dpkg` output

```
$ sudo dpkg -i logisim-evolution_3.9.0_amd64.deb
Selecting previously unselected package logisim-evolution.
(Reading database ... 304243 files and directories currently installed.)
Preparing to unpack logisim-evolution_3.9.0_amd64.deb ...
Unpacking logisim-evolution (3.9.0) ...
dpkg: dependency problems prevent configuration of logisim-evolution:
 logisim-evolution depends on libasound2t64; however:
  Package libasound2t64 is not installed.
 logisim-evolution depends on libpng16-16t64; however:
  Package libpng16-16t64 is not installed.

dpkg: error processing package logisim-evolution (--install):
 dependency problems - leaving unconfigured
Errors were encountered while processing:
 logisim-evolution
```

Installing via the graphical package manager (double clicking on the deb file in the file manager) also does not work and produces the error `Dependency resolution failed`.

### What I have tried

Attempting to call `apt --fix-broken install` removes the `logisim-evolution` package without doing anything else, and searching for the two packages in question finds that they don't exist, and the most similar packages (that is, they have the same name just different numbers at the end, i.e. `libasound2t64` doesn't exist, but `libasound2-dev` does) are already installed.

The `.jar` file works just fine, though I would expect that it would regardless.

Please let me know if you need more information.

maehne, 2024-09-15:
Both packages are present in Ubuntu 24.04 LTS, which we use to create the DEB package. Could you try to build Logisim-evolution on Debian 12 and check whether the created DEB package works? Enhancing our build pipeline to also consider this platform should not be too much work, but we currently lack the ressources to look into it, as the autumn term is about to start.

maehne, 2024-09-15:
That our DEB packages don't work on Debian has been already reported in issue #1569 for another reason. Therefore, I am marking this issue also as a duplicate.

maehne, 2024-09-15:
@MarcinOrlowski: Could you maybe have a look at this?

Animus-Surge, 2024-09-18:
> Could you try to build Logisim-evolution on Debian 12 and check whether the created DEB package works?

Did that, and it built just fine (though that's just from running from source). I'll create a deb package from `main` branch and see what happens

Animus-Surge, 2024-09-18:
Deb package built and installed without issue. Not an expert on packages so I wouldn't know the cause but it might be the required libraries are that slightly different to cause issues?

rail5, 2024-09-30:
The libraries provided by the **Ubuntu** packages `libasound2t64` and `libpng16-16t64` are provided in **Debian** by `libasound2` and `libpng16-16`.

Modifying the Depends: line in `debian/control` solves the problem:

```
Depends: libasound2t64 | libasound2, ... , libpng16-16t64 | libpng16-16, ...
```

I tested this on my local machine and the same Ubuntu-built binary installs and run just fine on my Debian 12

I would open a pull request but it looks like the Debian packaging isn't a part of the git repo(?) (I assume I'm just not seeing how it's handled)

rail5, 2024-09-30:
Makes sense -- I'm not familiar with that tool, but I would guess that it's checking which packages *on the build system* provide the libraries (which conversely would mean that if you built it on Debian, it wouldn't install on Ubuntu)

It's a shame that jPackage apparently doesn't yet support building any of the distro-agnostic package formats (AppImage, flatpak, etc)

Any solution I can think of would really just muddy up your build process -- so I'll just say thanks for maintaining such a great program, and probably most of us running Debian can figure it out

maehne, 2024-10-23:
I am going to close this issue, as we have issue #1569 to propose a DEB package compatible with Debian.

closed 2024-10-23 by maehne

tjcaul, 2025-03-01:
If you're looking for a workaround in the mean time, you can modify the `.deb` file yourself:

```
dpkg-deb -R logisim-evolution_*.deb logisim-evolution
sed -i 's/libasound2t64/libasound2/ ; s/libpng16-16t64/libpng16-16/' logisim-evolution/DEBIAN/control
dpkg-deb -b logisim-evolution logisin-evolution_fixed.deb
sudo dpkg -i logisin-evolution_fixed.deb
```