logisim-evolution/logisim-evolution #2067: Logisim evolution 1.9.0 проблема с изменением отображения схемы

ArturChuprina, 2024-09-27:
Когда я изменяю внешний вид схемы всё работает хорошо, после того как я сохраняю проект и выхожу из программы а после чего захожу в неё сново и открываю проект, внешний вид схемы сбрасывается тоесть вид пользовательский но положение контактов сбрасывается, как это исправить? Я планирую создать процессор и внешний вид очень важен, из-за этого создание процессора очень сложно. Я видел как другой человек задавал похожий вопрос и на него ответили но я не совсем понял как это решать

maehne, 2024-09-28:
Please state your problem in English. Only then, we might be able to help.

maehne, 2024-10-27:
Closing due to no reaction. Feel free to reopen if you can provide a concise description of your problem in English, which we can reproduce!

closed 2024-10-27 by maehne