logisim-evolution/logisim-evolution #2071: defect: [HDL] usage of >1 PortIO components

primechaser, 2024-10-04:
Hi Team,

the attached example circuit uses 2 Port IO components with 8 pins each.

The attached RTL (PIO initial.JPG, PIO initial_main.JPG) shows connector "bk0" has a bidirectional connection to the main circuit, whereas "bk1" is considered to be input only.
Additionally "bk0" is connected to internal logic, whereas "bk1" shows no connection at all.
Obviously this is not in line with the LE circuit.

The file "main_behavior.vhd" shows the following initial entries:
....
   -- Port I/O: bk0
    s_logisimBus5 <= logisimInOutBubbles(7 DOWNTO 0);
   logisimInOutBubbles(7 DOWNTO 0) <= s_logisimBus2 WHEN s_logisimNet0 = '1' ELSE (OTHERS => 'Z');

   -- Port I/O: bk1
    s_logisimBus1 <= logisimInOutBubbles(7 DOWNTO 0);
   logisimInOutBubbles(7 DOWNTO 0) <= s_logisimBus2 WHEN s_logisimNet0 = '1' ELSE (OTHERS => 'Z');
....

it looks as if "bk1" has not been set correctly.

after correction of "bk1" in "main_behavior.vhd" entry: 

   -- Port I/O: bk1
    s_logisimBus1 <= logisimInOutBubbles(7 DOWNTO 0);
    logisimInOutBubbles**(15 DOWNTO 8)** <= s_logisimBus2 WHEN s_logisimNet0 = '1' ELSE (OTHERS => 'Z');

connections appear to be correct: see RTL (PIO cor.JPG)

Thanks for looking in this

[PIO 2Units.zip](https://github.com/user-attachments/files/17256368/PIO.2Units.zip)

primechaser, 2024-10-08:
I missed to correct the input side, so for bk1 it should look like:

-- Port I/O: bk1
s_logisimBus1 <= logisimInOutBubbles**(15 DOWNTO 8)**;
logisimInOutBubbles**(15 DOWNTO 8)** <= s_logisimBus2 WHEN s_logisimNet0 = '1' ELSE (OTHERS => 'Z');

maehne, 2024-10-27:
I agree with you analysis and suggested manual fix to the netlist. PRs to fix this issue are welcome!

@BFH-ktt1: Could you maybe have a look what goes wrong?