logisim-evolution/logisim-evolution #2072: TTL 74181 is giving incorrect result when adding two numbers

craigmunday, 2024-10-06:
TTL 74181 is not adding numbers correctly.

A = 0x01
B = 0x01
No Carry

The result should be 0x02 but instead 0x01 is returned.

[image]

maehne, 2024-10-27:
Thanks for reporting this issue! A PR to fix this issue is welcome!

@BunchOfBits: Could you maybe have a look?

kevinjuszczyk, 2024-10-30:
This is not a bug. The function selected in the screenshot shown in the raised issue is A OR B, not A PLUS B. Below is a screenshot showing A PLUS B, selected with S0 = H, S1 = L, S2 = L and  S3 = H
[image]

maehne, 2025-01-03:
@kevinjuszczyk: Thanks for explaining that it is the expected behaviour. Closing this issue as resolved.

@craigmunday: Feel free to reopen in case of need for further discussion.

closed 2025-01-03 by maehne

raufie, 2025-04-13:
@craigmunday you're still right this is not the expected behavior. 
The datasheet clearly states that S3S2S1S0 should be, LLLH (only s0 should be high) for A+B, given M=0, Cn=1.
@kevinjuszczyk HLLH works but it doesn't follow the actual datasheet for 74181.

:( I guess I'm stuck with this one

[image]

kevinjuszczyk, 2025-04-13:
> The datasheet clearly states that S3S2S1S0 should be, LLLH (only s0 should be high) for A+B, given M=0, Cn=1. 

LLLH selects A+B, a Boolean algebra expression that means A OR B

HLLH selects A PLUS B, the arithmetic operation adding the numbers A and B.

IMO, the datasheet isn't clear, either due to a generation gap or abuse of notation.  ¯\_(ツ)_/¯

raufie, 2025-04-13:
Yeah @kevinjuszczyk , I also just found out lol... 
PLUS is add, + is OR.. unbelievable XD

Thanks

raufie, 2025-04-13:
A+B already exists in logic functions so I never imagined it'd be in arithmetic mode as well (M=L)

BunchOfBits, 2025-04-13:
Indeed, this can easily trip you up, the usage of + and PLUS in the datasheet. I implemented the 74181 in logisim, and have some actual chips laying around as well, which I used to check some questions I had myself. The function table in the datasheet contains accurate descriptions of what the chip does, which does not mean that all of them have any practical value: if you can come up with a practical use for A PLUS AB PLUS 1 (HLLL), then I'm really interested :-)