logisim-evolution/logisim-evolution #2076: Rendering issues on Windows

DevBlocky, 2024-10-13:
## Problem

When hovering over menus (or really anything with a hover animation), it will re-render stuff in the wrong place making the program completely unusable.

[Video](https://github.com/user-attachments/assets/952ee82d-2667-4c32-9d3e-d3631a2208cb)

Logisim-evolution will work correctly for a while, but suddenly it will starting doing the above. A restart will temporarily fix the issue, but eventually it will start again. Once it starts doing this, the only fix that I've found is saving and restarting the whole program. I've tried to look for similar issues on here, but haven't come across any.

## Troubleshooting I've Tried

1. Enabling or Disabling hardware acceleration
2. Installing and trying `logisim-evolution-3.9.0-x86_64.msi`
3. Trying `logisim-evolution-3.9.0-all.jar` on both java 8 and java 21

## Details
```
Product: Logisim-evolution v3.9.0
Runs on: Java HotSpot(TM) 64-Bit Server VM v21.0.4

Compiled: 2024-08-15T08:27:28+0200
Build ID: main/83c38fdb
Built on: Java HotSpot(TM) 64-Bit Server VM v21.0.4
```
OS: Windows 11

Let me know if you need anything additional

itzpr3d4t0r, 2024-10-18:
I've been having the exact same issue here on windows 11. But I suspect it might very well be related with the `RivaTuner Statistics Server` program as if i were to close that the issue would not present itself anymore.

davidhutchens, 2024-10-18:
@DevBlocky : Could you check to see if 1) you are using RivaTuner and 2) if disabling it resolves the issue for you?

adingler, 2024-10-18:
For what it's worth, I also have this issue (on Win11), and I do _not_ have RivaTuner.  At least one of my students reported this issue on a Mac, but I don't know which OS version and cannot replicate on my Mac.

XX-IX-II-II-V, 2024-10-20:
I got the same problem. I also do have Rivatuner but it isn't running.

davidhutchens, 2024-10-23:
The latest Flatlaf release (which was just merged into our main branch) has the following release note: "Windows: Fixed repaint issues (ghosting) on some systems (probably depending on graphics card/driver). This is done by setting Java system property sun.java2d.d3d.onscreen to false (but only if sun.java2d.d3d.onscreen, sun.java2d.d3d and sun.java2d.noddraw are not yet set), which disables usage of Windows Direct3D (DirectX) onscreen surfaces. Component rendering still uses Direct3D. (issue [#887](https://redirect.github.com/JFormDesigner/FlatLaf/issues/887))"

@DevBlocky, @itzpr3d4t0r, @adingler, @XX-IX-II-II-V: It may be a long shot, but could those who have seen this issue try our latest [Nightly build](https://github.com/logisim-evolution/logisim-evolution/actions/workflows/nightly.yml) to see if the problem still remains? Please report your findings here.

DevBlocky, 2024-10-30:
> It may be a long shot, but could those who have seen this issue try our latest [Nightly build](https://github.com/logisim-evolution/logisim-evolution/actions/workflows/nightly.yml) to see if the problem still remains? Please report your findings here.

@davidhutchens I'm pretty sure this fixed the issue! I've used it for about an hour now, which is much longer than I could before 😄

davidhutchens, 2024-10-31:
@DevBlocky That is good news. Since this was your issue, when you are confident it is fixed, you can close it as completed. It would also be reasonable to wait a few more days to see if anyone else wishes to report their findings. In either case we can always reopen it later if the issue reappears.

adingler, 2024-11-01:
I tried this out in class today and during the 20 minutes we were building a finite state machine, it did not glitch once.  Usually it would happen within a couple minutes.  Thanks!

maehne, 2024-11-11:
@DevBlocky: We have not yet discussed a target date for a new stable release. From our side, a release before the start of the spring term would make sense. @BFH-ktt1: What do you think?