logisim-evolution/logisim-evolution #2089: Build (Java OpenJDK 23, gradle 8.10.2) fails with `VhdlSyntax.java:42: error: VhdlSyntax is not abstract and does not override abstract method yystate() in AbstractJFlexTokenMaker`.

dreirund, 2024-11-02:
Ahoj,

I am trying to build this (with gradle version 8.10.2, Java version openjdk 23 2024-09-17, on [Artix GNU/Linux](http://artixlinux.org/) (rolliing release)).

Build step (`gradle --no-daemon shadowJar`) fails with `VhdlSyntax.java:42: error: VhdlSyntax is not abstract and does not override abstract method yystate() in AbstractJFlexTokenMaker`:  
```
> Task :compileJava
/var/cache/makepkg/build/logisim-evolution-git/src/logisim-evolution/build/generated/logisim/java/com/cburch/logisim/vhdl/syntax/VhdlSyntax.java:42: error: VhdlSyntax is not abstract and does not override abstract method yystate() in AbstractJFlexTokenMaker
public class VhdlSyntax extends AbstractJFlexTokenMaker {
       ^
1 error

> Task :compileJava FAILED

FAILURE: Build failed with an exception.

* What went wrong:
Execution failed for task ':compileJava'.
> Compilation failed; see the compiler error output for details.

* Try:
> Run with --info option to get more log output.
> Run with --scan to get full insights.

BUILD FAILED in 10s
3 actionable tasks: 1 executed, 2 up-to-date
```

Adding `--info --scan` to the `gradle` command:  
```
> Task :compileJava
Caching disabled for task ':compileJava' because:
  Build cache is disabled
Task ':compileJava' is not up-to-date because:
  Task has failed previously.
The input changes require a full rebuild for incremental task ':compileJava'.
Compilation mode: in-process compilation
Full recompilation is required because no incremental change information is available. This is usually caused by clean builds or changing compiler arguments.
Compiling with toolchain '/usr/lib/jvm/java-23-openjdk'.
file or directory '/var/cache/makepkg/build/logisim-evolution-git/src/logisim-evolution/build/generated/sources/srcgen', not found
Compiling with JDK Java compiler API.
/var/cache/makepkg/build/logisim-evolution-git/src/logisim-evolution/build/generated/logisim/java/com/cburch/logisim/vhdl/syntax/VhdlSyntax.java:42: error: VhdlSyntax is not abstract and does not override abstract method yystate() in AbstractJFlexTokenMaker
public class VhdlSyntax extends AbstractJFlexTokenMaker {
       ^
1 error

> Task :compileJava FAILED

FAILURE: Build failed with an exception.

* What went wrong:
Execution failed for task ':compileJava'.
> Compilation failed; see the compiler error output for details.

BUILD FAILED in 28s
3 actionable tasks: 3 executed
Some of the file system contents retained in the virtual file system are on file systems that Gradle doesn't support watching. The relevant state was discarded to ensure changes to these locations are properly detected. You can override this by explicitly enabling file system watching.
```

Regards!

davidhutchens, 2024-11-02:
This is a duplicate of issue #2088. You can fix it in your build by changing the rsyntaxtextarea reference in build.gradle.kits to be 3.5.1 rather than 3.5.2.

maehne, 2024-11-03:
Closing as duplicate.

closed 2024-11-03 by maehne

dreirund, 2024-11-03:
> *This is a duplicate of issue https://github.com/logisim-evolution/logisim-evolution/issues/2088. You can fix it in your build by changing the rsyntaxtextarea reference in build.gradle.kits to be 3.5.1 rather than 3.5.2.* 

Applying the patch [`https://github.com/logisim-evolution/logisim-evolution/pull/2084.patch`](https://github.com/logisim-evolution/logisim-evolution/pull/[2084.diff](https://github.com/logisim-evolution/logisim-evolution/pull/2084.patch)) reversed solves this problem, yes.

Thanks!