logisim-evolution/logisim-evolution #2099: Error (suppressible): (vcom-19) Failed to access library 'work' at "work".

Shanksss0500, 2024-11-22:
Just by opening my Logisim Evolution, this error pops up on the screen, and I don’t know how to fix it.
[image]

maehne, 2024-11-22:
Have you maybe checked "Use Questa Advanced Simulator to validate HDL entities" in Window -> Preferences -> Software? If yes, please disable that option and restart Logisim-evolution to check whether the issue persists. If this doesn't fix the issue, we need more precise information on how to reproduce the issue step by step on our side. This should include the exact version of Logisim-evolution and detailed information on which OS platform you are using it and how Logisim-evolution settings have been altered from their defaults. An example design in form of a CIRC-file inside a ZIP archive attached to a message alongside step-by-step instructions would be also helpful.

Please note that support for co-simulating VHDL entities as part of a Logisim-evolution circuit is unfortunately not mature and none of the currently active maintainers is using/working on it.

PRs to improve the situation are welcome!

BFH-ktt1, 2025-05-27:
Closing issue as no follow up, please reopen if the problem persists.