logisim-evolution/logisim-evolution #2101: VHDL under Linux Mint

Cuvy95, 2024-11-24:
I trie to create a simple VHDL File, but every time i want to run the program, i got the message "Simulation aufgrund eines internen Fehler angehalten".
My VHDL Code looks like:

library IEEE;
use IEEE.STD_LOGIC_1164.ALL;

-- Entity Definition
entity AND_Gate is
    Port (
        A : in  STD_LOGIC; -- Eingang A
        B : in  STD_LOGIC; -- Eingang B
        Y : out STD_LOGIC  -- Ausgang Y
    );
end AND_Gate;

-- Architecture Definition
architecture Behavioral of AND_Gate is
begin
    -- Logik für das AND-Gatter
    Y <= A and B;
end Behavioral;

When i try the same code in "digital" the vhdl code run fine and i can use the VHDL "and" and it worked as an And-Block.
[image]

maehne, 2024-11-25:
Have set up the path to ModelSIM/Questa Sim in `Window` -> `Preferences` -> `Software` -> `Questa Advanced Simulator path`? E.g., on my laptop, the path is `/opt/intelFPGA/20.1/modelsim_ase/bin`. Then, you have to make sure that `Simulate` -> `VHDL Simulation Enabled` is checked. In case of an internal error, you may need to select `Restart VHDL Simulator` and `Reset Simulation` in the `Simulate` menu as well. Then, your simple test case works on my machine. In your screenshot, the `and_gate` is not properly placed though.

We are aware that the co-simulation of VHDL models as part of Logisim-evolution has many restrictions and works not in all cases flawlessly. However, that code part was contributed by people, who haven't been actively involved in Logisim-evolution's development for some time now. As we are not actively using this feature in our teaching, we cannot devote ressources to it. PRs to improve the functionality and add support for free VHDL/Verilog simulatorsare very welcome though!

Cuvy95, 2024-11-26:
Thanks for your help, now i have a better understanding for the problem.
I have now installed the Intel ModelSim Starter Edition and i have copied the path into Logisim.
But now i have this error "Error: cannot find "/home/cuvy/intelFPGA/18.1/modelsim_ase/bin/../linux_rh60/vlib".
But in the installation path this folder is not there. What did i do wrong at the installation?

BFH-ktt1, 2025-05-27:
closing issue as no follow up. Please reopen if you feel the problem persists.

closed 2025-05-27 by BFH-ktt1