logisim-evolution/logisim-evolution #2103: Generic autosave error if opened file doesn't end with .circ extension

zdimension, 2024-11-29:
Somehow, I ended up having a circuit file that doesn't have an extension (maybe because the name was mistyped, I don't know).

Opening it works, but after a while, the autosave is triggered, here:

https://github.com/logisim-evolution/logisim-evolution/blob/db7c97cab2e51e1dde0202eddd7c3e3621be28d4/src/main/java/com/cburch/logisim/file/LogisimFile.java#L96-L101

This calls the following:

https://github.com/logisim-evolution/logisim-evolution/blob/db7c97cab2e51e1dde0202eddd7c3e3621be28d4/src/main/java/com/cburch/logisim/file/Loader.java#L153-L166

Line 162 returns null if the file doesn't have the proper extension, which triggers a generic autosave error message in the first snippet. Either a better error message should be displayed, or the autosave code should be fixed to handle extensionless files.

maehne, 2024-12-06:
I agree that the error message should be more precise when the correct file extension is missing. A PR in this direction and also with additional enhancements to deal with this issue is welcome!

zdimension, 2024-12-07:
Is there any reason why we wouldn't want to autosave extensionless files? @superenginegit (tagging since you wrote the autosave code)

https://github.com/logisim-evolution/logisim-evolution/blob/db7c97cab2e51e1dde0202eddd7c3e3621be28d4/src/main/java/com/cburch/logisim/file/Loader.java#L162

I mean, what would be wrong in creating an autosave file `foobar.autosave` if the currently opened file is `foobar`? Is there a breaking case I'm not thinking about?

superenginegit, 2024-12-07:
I think it should work fine to just remove this line. I am not sure why I implemented it like that, I most likely didn't consider this case ever happening.

It would also be a possible to simply add a .circ.autosave extension for the autosave, something like this should work:
```java
    var extension = ".autosave";
    if (!base.getName().endsWith(LOGISIM_EXTENSION)) extension = ".circ.autosave";
    final var dir = base.getParentFile();
    final var name = "." + base.getName() + extension;
    return new File(dir, name);
```

maehne, 2024-12-12:
I like the suggestion by @superenginegit to add the missing `.circ` extension at least to the autosave file. Maybe it would be also good to show a message informing the user about the missing extension. A PR is welcome!

closed 2025-05-03 by BFH-ktt1