logisim-evolution/logisim-evolution #2106: Doesn't support german Charackters

Nino678190, 2024-12-03:
The App crashes if you try to type Ä,Ü,Ö,ß

maehne, 2024-12-06:
I cannot reproduce this issue. Please, be more precise by explaining step by step how to reproduce the issue! Also provide additional information about your environment! What exact version of Logisim-evolution are you using? Can you reproduce the issue with the latest official release and the latest nightly release? Which operating system are you using? On which processor platform?

Please also note that Logisim-evolution restricts on purpose the set of allowed characters for any labels in a schematic to ensure that the labels are always legal VHDL and Verilog identifiers. This is required for enabling the export for logic synthesis by a 3rd-party FPGA toolchain.

Nino678190, 2024-12-08:
Used a old version from the snap store with 3.8.0

closed 2024-12-08 by Nino678190