logisim-evolution/logisim-evolution #2110: Simulator crashes on non-50% clock duty cycle

hawaclawek, 2024-12-10:
v3.8.0. Directions for reproduction on a blank schematic:
 - Place "Wiring - Clock"
 - Leave "High Duration" on 1 Tick
 - Set "Low Duration" to 2 Ticks
 - "Simulate - Timing Diagram"
 - "Timing Diagram - Enable Clock Ticks"
 --> "The simulator has crashed. Save your work and restart Logisim."

maehne, 2025-01-03:
Sorry, I cannot reproduce the issue with the current development HEAD. Could you please test with the [latest nightly release](https://github.com/logisim-evolution/logisim-evolution#nightly-builds)? Also, it would be nice if you could provide a test case in form of a CIRC file in a ZIP archive as well as a screencast showing the occurence of the issue.

Please also provide addition details on your platform. Do you have by any chance Questa Sim/ModelSIM configured/active for VHDL co-simulation?

hawaclawek, 2025-01-03:
Issue reproduced using nightly build _logisim-evolution_main_20241224_amd64.deb_. Please find the circ file and screencast documenting the reproduction attempt attached.
[rep_issue_2110.zip](https://github.com/user-attachments/files/18302570/rep_issue_2110.zip)

maehne, 2025-01-06:
Thanks a lot for the reproducer! I am now able to reproduce the issue. It is interesting that the crash only gets triggered when enabling the clock ticks from *within* the Timing Diagram window. In it, it doesn't matter whether the icons or the Simulate menu is used.

@BFH-ktt1: Could you have a look?

Diogo-Valadares, 2025-01-22:
I feel like its something to do with the Timing Diagram not inicializing the timeline correctly. 

It seems like when you open it after the clock has been created, it starts with 5000ns which is fine if the low is only 1 tick, but if the clock low has 2 or more ticks, the low duration will exceed those first 5000ns, triggering this exception:
 "signal data should be at least 20000 ns in duration, but only 5000 in last signal at com.cburch.logisim.gui.log.Signal.replaceRecent(Signal.java:143)"

After reseting the simulation the timeline seems to be correctly reseted and the problem no longer happens, unless you increase the low ticks again.
[image]
[image]

> It is interesting that the crash only gets triggered when enabling the clock ticks from within the Timing Diagram window. 

It did happened to me with the enable clock ticks button outside the Timing Diagram, but having the Timing Diagram open seems to be a requirement.

I tried to remove the exception([lines 143-149 of the Signal.java](https://github.com/logisim-evolution/logisim-evolution/blob/main/src/main/java/com/cburch/logisim/gui/log/Signal.java#L143-L149)) just to see if it would break anything and it didn't, but I think it would need a deeper investigation to find a better fix.