logisim-evolution/logisim-evolution #2117: [bug] I/O resource `SevenSegmentScanning` cannot be synthesized due to missing info in constraints file

berryboy2012, 2024-12-29:
As stated in the title, the following circuit will not work when seven segment displays are mapped on FPGA boards' `SevenSegmentScanning` entries. I have only tested this on vivado boards, but I suspect that boards from other vendors have similar issue as I went through the code responsible for generating constraint files.
[image]

## Environment
OS: Windows 11

Software version: 3.9.0

## Steps to reproduce
1. Create the circuit shown above
2. Click `FPGA - Synthesize and Download ...`
3. Select `ALCHITRY_AU_IO` as target board
4. Proceed with settings and hit `Execute`
5. Observe console output and then the generated `vivadoConstraints.xdc` file

maehne, 2025-01-03:
This may be related to PR #1798 by @BFH-ktt1. Support for the alchitry au io board was added by @chuckb in PR #1648. We don't use neither Vivado nor the Alchitry board in our teaching, as we currently primarily use Intel/Altera-based FPGA boards. It would be helpful if you could provide a test case making use of all seven segment displays on the board together with

- the full output of the Vivado tool flow as shown in the Console tab of the Synthesize & Download dialogue

@chuckb and @BFH-ktt1: Could you maybe have a look?

berryboy2012, 2025-01-08:
Sorry kind of busy lately. The problem can be reproduced by simply going through the synthesis and implementation stages, there's no need to own an actual Xilinx FPGA to observe the error. With that being said, I'll try to provide the necessary logs once sooner or later.