logisim-evolution/logisim-evolution #2123: How to get VHDl to work?

rafal11ck, 2025-01-07:
I'am trying to learn VHDL.

- I used `add VHDL` option in top left.
- Made code as following (just NOT gate)
```
LIBRARY ieee;
USE ieee.std_logic_1164.all;

ENTITY test IS
  PORT (
    x      : IN  std_logic;                    -- input bit example
    y        : OUT std_logic                    -- output bit example
    );
END test;

ARCHITECTURE TypeArchitecture OF test IS
BEGIN
    y <= not x;
END TypeArchitecture;
```

I except it to work, instead got blue line
[image]

I went to docs didn't find anything related to VHDL.

searched for existing issues and found 
#2101 and #1966 .

As mentioned in one of them I go to `window->preferences->software`
[image]

- What are those? Do I need all of them, one or subset?
- Getting some kind of error, instead of `internal error` after trying to restart simulation would be nice.

---

After I wrote that, I found this in terminal
```
java.lang.UnsupportedOperationException: VHDL component simulation is not supported. This could be because there is no Questasim/Modelsim simulation server running.
        at com.cburch.logisim.vhdl.base.VhdlEntity.propagate(VhdlEntity.java:253)
        at com.cburch.logisim.instance.InstanceComponent.propagate(InstanceComponent.java:414)
        at com.cburch.logisim.circuit.CircuitState.processDirtyComponents(CircuitState.java:341)
        at com.cburch.logisim.circuit.Propagator.propagate(Propagator.java:275)
        at com.cburch.logisim.circuit.Simulator$SimThread.loop(Simulator.java:474)
        at com.cburch.logisim.circuit.Simulator$SimThread.run(Simulator.java:538)
```
This might be answer to first question, maybe show that instead error along with internal error text?

maehne, 2025-01-16:
Indeed, only https://github.com/logisim-evolution/logisim-evolution/issues/2101 and https://github.com/logisim-evolution/logisim-evolution/issues/1966 provide currrently the information about setting up ModelSim / Questa Sim to be able to co-simulate VHDL models within a Logisim circuit. So, you just need to install the tool and set up the path in the first entry. The other paths are for FPGA toolchains to enable synthesis for different FPGA families using their respective vendor tool chain.

PRs to improve the documentation in this respect are welcome. We don't use/advertise VHDL co-simulation in our teaching and currently don't have resources to work on it.

crossplatformdev, 2025-01-25:
Here are the instructions for the old Logisim classic, in Spanish, but you can use the subtitles.

If I get a **compatible** FPGA I will make my own bilingual video. Anyone does know which one is affordable and can reach above 100Mhz?

https://www.youtube.com/watch?v=U5cp65uaUzw

maehne, 2025-02-13:
@crossplatformdev: An FPGA board is not required to do VHDL simulation. The ModelSim / Questa Sim versions offered together with Quartus Prime lite are also offered as an independent download. It is difficult to give good recommendation for an FPGA board with just the constraint that it "can reach above 100MHz". Most FPGAs offer integrated an integrated PLL to generate higher frequencies from the clock signal provided by the quartz on the PCB.

I recommend that you clarify your actual requirements by thinking for the kind of applications you would like to use it for. By choosing the FPGA type, you also buy into an ecosystem with a particular toolchain and advantages/disadvantages. Logisim-evolution supports FPGAs from AMD/Xilinx, Intel/Altera (both using the vendor's tool chain available for Windows and Linux), and Lattice (currently only tested with ECP5 using the free toolchain based on Yosys and nextPnR). The list of supported FPGA boards in the "Synthesize & Download" dialog can serve as a starting point. However, some boards are already starting to get old featuring older FPGA families with growing uncertainty how long they will remain actively supported in the vendor's most current tool chain versions. E.g., Xilinx Vivado only offers support for Series 7 FPGAs and newer. If none of the supported boards suite you, you can add support for a new board with reasonable effort as long as the integrated FPGA is supported by the tool chains used by Logisim-evolution.

Mudassir10X, 2025-04-26:
# Just give the PATH to the simulator
You just need one of the simulators but have to tell the Logisim which one and where is the exe or binary for the simulator that you want to use as the Logisim is not a simulator for vhdl. Now can we mark this issue as closed?

closed 2025-04-26 by rafal11ck