logisim-evolution/logisim-evolution #2136: Xilinx ISE Webpack compatibility

Phifo31, 2025-02-06:
Dear developers, first I'd like to thank you for the amazing work you are doing developing Logisim. Second, I would like to ask some questions but first, I want to expose my problem.

I use Logisim as a teaching tool with my students. I use Vidor 4000 with a cyclone LP10 and Quartus tools. All is clear and works well with Logisim evolution.
You can find my XML File for Vidor 4000 on MKR Connector carrier here : https://github.com/Phifo31/vidor4logisim

I also have old FPGA boards from Digilent named NEXYS3 with Spartan 6 and I use them with Xilinx ISE Webpack (alone, without logisim).
But, When I try to use "FPGA synthetize and download" with a board like Nexys3 or Basys2, using Xilinx old tools (ISE Webpack) I have this error and logisim freeze :

Exception in thread "Thread-2" java.lang.RuntimeException: #E006: No mapping for 'JTAGPos' placeholder in 'assignFile -p {{JTAGPos}} -file {{fileBaseName}}.{{fileExt}}
program -p {{JTAGPos}} -onlyFpga
'.
	at com.cburch.logisim.util.LineBuffer.abort(LineBuffer.java:685)
	at com.cburch.logisim.util.LineBuffer.abort(LineBuffer.java:695)
	at com.cburch.logisim.util.LineBuffer.validateLine(LineBuffer.java:820)
	at com.cburch.logisim.util.LineBuffer.validateLineNoPositionals(LineBuffer.java:731)
	at com.cburch.logisim.util.LineBuffer.add(LineBuffer.java:281)
	at com.cburch.logisim.util.LineBuffer.add(LineBuffer.java:265)
	at com.cburch.logisim.fpga.download.XilinxDownload.createDownloadScripts(XilinxDownload.java:232)
	at com.cburch.logisim.fpga.download.Download.createDownloadScripts(Download.java:200)
	at com.cburch.logisim.fpga.download.Download.prepareDownload(Download.java:408)
	at com.cburch.logisim.fpga.download.Download.run(Download.java:220)
	at java.base/java.lang.Thread.run(Thread.java:1575)

To reproduce : create schematic with a not gate, one input and one output, configure tools for xilinx, annotate, select pins with the graphic tool and push the "Done" button.

I never tried Vivado because the Spartan6 family is not supported.

I use Logisim v3.8.0 and 3.9.0, java-23-openjdk, Xilinx ISE WebPack v, and archlinux up-to-date.

Strangely, the old version logisim-evolution 2.15 seems to work but is not compatible with our actual work.

Could you give me information about the compatibility ?
Is the problem only with my configurations ?
Is it possible to use a generic tool with special scripting (bash) to pass this error ?
Do you have some tips to try ?

Thanks again for your works
BR,
Phifo 31

maehne, 2025-02-12:
Thanks for your positive feedback and reporting the Xilinx ISE issue @Phifo31! 

You can create a pull request to add the Vidor 4000 XML file in `src/main/resources/resources/logisim/boards/` so that it becomes part of the list of directly supported FPGA boards by Logisim-evolution.

Unfortunately, we don't have anymore access to such old Xilinx FPGA boards to debug this error ourselves. In addition, Xilinx ISE isn't supported by AMD/Xilinx since many years and has become difficult to install on recent OS versions requiring not officially supported/documented patching of the installed tool chain. Therefore, we cannot support you in finding the cause of the issue. A good starting point is searching for `JTAGPos` in the Logisim-evolution code base. Besides the board XML files, it is referecenced in four Java files, which limits the search:

- `FpgaClass.java`
- (`VivadoDownload.java`)
- **`XilinxDownload.java`**
- `BoardWriterClass.java`

The error message is generated in `LineBuffer.java`, which class is used in `XilinxDownload.java`.

Maybe, @BFH-ktt1 can provide further hints to limit the search.

Logisim-evolution currently does not include an XML file for `NEXYS3`. Feel free to make a PR adding it once the above issue is resolved.

bonze82, 2025-04-19:
I have 2 ztex boards with 4x spartan6 lx150's each. The latest ise 14.6 will install on debian and is relatively bug free, using right now to port python code to verilog for a project of mine.

maehne, 2025-04-24:
@bonze82: Thanks for this feedback!

PRs to keep Logisim-evolution compatible with Xilinx ISE and adding support for new FPGA boards are always welcome!