logisim-evolution/logisim-evolution #2138: Does not select bits lower than already selected

SalonJee, 2025-02-08:
when i select bit 3 , if i type 2 and hit enter, it autmatically selects 20 bits , instead of 2. It only works if i had already chosen lower number of bits ( say 1) , only then it selects 2 when i type 2 . 

OS - Windows 11
Software version : 2.7.1

juli-99, 2025-02-08:
2.7.1 was the last version from the original Logisim (2011).
Try using [the latest version of Logisim-evolution](https://github.com/logisim-evolution/logisim-evolution/releases).

maehne, 2025-02-12:
Closing this, as we don't have the resources to support outdated versions of Logisim. If the issue persists in the current official or nightly release, please reopen the issue and provide more detailed instructions how to reproduce the issue. The above description is too vague. If the issue should be difficult to explain in words, you can add screenshots and/or a short screencast illustrating the issue.