logisim-evolution/logisim-evolution #2144: Gradle: Deprecation warning "Invocation of Task.project at execution time has been deprecated."

maehne, 2025-02-12:
Building Logisim-evolution from HEAD yields the following deprecation warning:

```
$ ./gradlew --warning-mode all createDmg

> Task :createDmg
Invocation of Task.project at execution time has been deprecated. This will fail with an error in Gradle 10.0. This API is incompatible with the configuration cache, which will become the only mode supported by Gradle in a future release. Consult the upgrading guide for further information: https://docs.gradle.org/8.12.1/userguide/upgrading_version_7.html#task_project
        at Build_gradle$13$2.execute(build.gradle.kts:515)
        (Run with --stacktrace to get the full stack trace of this deprecation warning.)

[Incubating] Problems report is available at: file:///Users/maehne/Build/logisim-evolution/build/reports/problems/problems-report.html
```

The referenced HTML report contains the same deprecation message. The [Gradle upgrading guide](https://docs.gradle.org/8.12.1/userguide/upgrading_version_7.html#task_project) and [Configuration cache documentation](https://docs.gradle.org/8.12.1/userguide/configuration_cache.html#config_cache:requirements:use_project_during_execution) contain hints, what needs to be done. Indeed, we reference the properties `project.name`, `project.version` etc. in many places of `build.gradle.kts`.

@davidhutchens: Could you maybe have a look?

davidhutchens, 2025-02-12:
I will look into it. At first glance it looks like we need to make all the project.something references into constants so we do not invoke project at runtime. It doesn't seem to have complained about the uses in extra.apply definitions.

@maehne: Could you check if there are warnings on LInux or Windows when using --warning-mode all?

maehne, 2025-02-13:
@davidhutchens: Thanks that you are willing to look into this issue. On Linux, I am not observing the deprecation warning:

```
$ ./gradlew --warning-mode=all createDeb

BUILD SUCCESSFUL in 48s
7 actionable tasks: 7 executed
$ ./gradlew --warning-mode=all clean    

BUILD SUCCESSFUL in 853ms
1 actionable task: 1 executed
$ ./gradlew --warning-mode=all createRpm

BUILD SUCCESSFUL in 49s
7 actionable tasks: 7 executed
```

I will check on Windows once I get around to it.

maehne, 2025-02-13:
On Windows, I get the deprecation warning:

```
 ./gradlew --warning-mode=all createMsi

> Task :createMsi
Invocation of Task.project at execution time has been deprecated. This will fail with an error in Gradle 10.0. This API is incompatible with the configuration cache, which will become the only mode supported by Gradle in a future release. Consult the upgrading guide for further information: https://docs.gradle.org/8.12.1/userguide/upgrading_version_7.html#task_project
        at Build_gradle$11$2.execute(build.gradle.kts:394)
        (Run with --stacktrace to get the full stack trace of this deprecation warning.)

[Incubating] Problems report is available at: file:///C:/Users/mht1/Build/logisim-evolution/build/reports/problems/problems-report.html

BUILD SUCCESSFUL in 1m 49s
7 actionable tasks: 7 executed
```

davidhutchens, 2025-02-16:
Thank you, @maehne, for the Linux and Windows results. They helped make clear what I needed to do.

A PR to fix this will follow.

davidhutchens, 2025-02-16:
Completed with #2148.