logisim-evolution/logisim-evolution #2145: Exception java.lang.IndexOutOfBoundsException: Index 0 out of bounds for length 0 [in thread "main"]

mnp555, 2025-02-12:
I'm trying compile project in Itellij IDEA everything works fine.
When i launch java -jar logisim-evolution.jar i get:

Exception in thread "main" Exception in thread "Event Dispatch Thread" java.lang.NoClassDefFoundError: Could not initialize class com.cburch.logisim.prefs.AppPreferences
        at com.cburch.logisim.prefs.AppPreferences$MyListener.preferenceChange(AppPreferences.java:117)
        at java.prefs/java.util.prefs.AbstractPreferences$EventDispatchThread.run(AbstractPreferences.java:1539)
Caused by: java.lang.ExceptionInInitializerError: Exception java.lang.IndexOutOfBoundsException: Index 0 out of bounds for length 0 [in thread "main"]
        at java.base/jdk.internal.util.Preconditions.outOfBounds(Preconditions.java:100)
        at java.base/jdk.internal.util.Preconditions.outOfBoundsCheckIndex(Preconditions.java:106)
        at java.base/jdk.internal.util.Preconditions.checkIndex(Preconditions.java:302)
        at java.base/java.util.Objects.checkIndex(Objects.java:385)
        at java.base/java.util.ArrayList.get(ArrayList.java:427)
        at com.cburch.logisim.prefs.FpgaBoards.<init>(FpgaBoards.java:213)
        at com.cburch.logisim.prefs.AppPreferences.<clinit>(AppPreferences.java:505)
        at com.cburch.logisim.Main.main(Main.java:43)
java.lang.ExceptionInInitializerError
        at com.cburch.logisim.Main.main(Main.java:43)
Caused by: java.lang.IndexOutOfBoundsException: Index 0 out of bounds for length 0
        at java.base/jdk.internal.util.Preconditions.outOfBounds(Preconditions.java:100)
        at java.base/jdk.internal.util.Preconditions.outOfBoundsCheckIndex(Preconditions.java:106)
        at java.base/jdk.internal.util.Preconditions.checkIndex(Preconditions.java:302)
        at java.base/java.util.Objects.checkIndex(Objects.java:385)
        at java.base/java.util.ArrayList.get(ArrayList.java:427)
        at com.cburch.logisim.prefs.FpgaBoards.<init>(FpgaBoards.java:213)
        at com.cburch.logisim.prefs.AppPreferences.<clinit>(AppPreferences.java:505)

Any idea.
please some help I'm not experienced coder :).
thx

maehne, 2025-02-13:
Your issue description is too vague to diagnose the problem.

Please check if the issue persists when building Logisim-evolution from the command line using Gradle, as described [here](https://github.com/logisim-evolution/logisim-evolution/blob/main/docs/developers.md). If yes, please provide the exact sequence of commands, which you use to build/run Logisim-evolution. Also specify, which state of the sources (i.e., Git commit), which exact JDK version, operating system version, and processor platform you are using.

mnp555, 2025-02-14:
My machine: win 11, ram 64g, i9 9900k 

Using gradle i get:

* What went wrong:
A problem occurred configuring root project 'logisim-evolution'.
> Could not resolve all artifacts for configuration 'classpath'.
   > Could not resolve org.sonarsource.scanner.gradle:sonarqube-gradle-plugin:6.0.1.5171.
     Required by:
         root project : > org.sonarqube:org.sonarqube.gradle.plugin:6.0.1.5171
      > Dependency requires at least JVM runtime version 11. This build uses a Java 8 JVM.

Workflow in IntellijIDE:
1) opening project same gradle output
2) in Settings --> Build, Exec .. --> Build Tools --> Gradle --> Gradle JVM set Amazon Corretto11.0.26 
3) closing and reopening IDE, everything OK
4) Project Structure --> Project Settings --> Project --> SDK --> Corretto11.0.26 with language level 11
5) Project Structure --> Project Settings --> Modules --> language level 11
6) Project Structure --> Project Settings --> Platform Settings SDKs --> Corretto11.0.26
7) Build Project --> give error:
	* What went wrong:
Execution failed for task ':compileJava'.
> error: invalid source release: 21
8) redo step 2 to 6 setting openjdk-21 Oracle and language level 21
9) Build project OK
10) Project Structure --> Project Settings --> Artifacts add JAR Module logisim-evolution.main, main class com.cburch.logisim.Main, extract to the target JAR and setting correct directory for MANIFEST.FS
11) Build Artifacts  --> logisim-evolution.jar -->  Build  OK
12) in directory out/artifacts/logisim_evolution_main_jar i have the file logisim-evolution.main.jar
13) opening the location in terminal and running java -jar logisim-evolution.main.jar, now I get:

Exception in thread "main" Exception in thread "Event Dispatch Thread" java.lang.ExceptionInInitializerError
        at com.cburch.logisim.util.LocaleManager.setLocale(LocaleManager.java:197)
        at com.cburch.logisim.prefs.AppPreferences$LocalePreference.<init>(AppPreferences.java:85)
        at com.cburch.logisim.prefs.AppPreferences.<clinit>(AppPreferences.java:489)
        at com.cburch.logisim.Main.main(Main.java:43)
Caused by: java.lang.RuntimeException: No locale bundles are available
        at com.cburch.logisim.util.LocaleManager.loadDefault(LocaleManager.java:328)
        at com.cburch.logisim.util.LocaleManager.<init>(LocaleManager.java:103)
        at com.cburch.logisim.util.Strings.<clinit>(Strings.java:18)
        ... 4 more
java.lang.NoClassDefFoundError: Could not initialize class com.cburch.logisim.prefs.AppPreferences
        at com.cburch.logisim.prefs.AppPreferences$MyListener.preferenceChange(AppPreferences.java:117)
        at java.prefs/java.util.prefs.AbstractPreferences$EventDispatchThread.run(AbstractPreferences.java:1539)
Caused by: java.lang.ExceptionInInitializerError: Exception java.lang.ExceptionInInitializerError [in thread "main"]
        at com.cburch.logisim.util.LocaleManager.setLocale(LocaleManager.java:197)
        at com.cburch.logisim.prefs.AppPreferences$LocalePreference.<init>(AppPreferences.java:85)
        at com.cburch.logisim.prefs.AppPreferences.<clinit>(AppPreferences.java:489)
        at com.cburch.logisim.Main.main(Main.java:43)

it's all :)

maehne, 2025-02-14:
So you have several JDKs installed in parallel on your Windows installation. Logisim-evolution requires JDK 21 or newer. Make sure to [switch to the correct JDK](https://stackoverflow.com/questions/26993101/switching-between-different-jdk-versions-in-windows) by setting `JAVA_HOME` and `PATH` correctly. Check with `javac -version` in a newly opened terminal that indeed the right Java compiler version is used for compilation. Then, build Logisim-evolution with Gradle from that command line.: `./cradlew runShadow`. Only when this yields the same error, we may be able to reproduce it on our side.

For the moment, we are not able to reproduce your issue locally.

We won’t be able to assist you with resolving IDE-specific issues or those related to a mixed setup of several installed JDKs.

mnp555, 2025-02-15:
I correct setup of JAVA_HOME and PATH and everything works.
Thx for helping

closed 2025-02-15 by maehne