logisim-evolution/logisim-evolution #2153: Fix synchronization issues in CircuitWires, CircuitState, and Canvas

davidhutchens, 2025-02-17:
We should incorporate the synchronization changes and fixes made by @kevinawalsh in CircuitWires, CircuitState, and Canvas. It may change other classes and files as well.

@BFH-ktt1: I have now looked into these twice for #1805 and #2129. I know you were planning to work on this, but if you would like, I am willing to give it a try. Please let me know. I don't want us to both work on it and waste effort.

BFH-ktt1, 2025-02-18:
@davidhutchens : It has already been a while that I did not have time to work on logisim, and I foresee in the coming weeks that I will also not be able to work on it. So please go ahead.

davidhutchens, 2025-03-22:
PR #2170 contains most of the changes for this issue. It does not contain the changes to Canvas. I'll work on that next.

davidhutchens, 2025-04-06:
With the merger of PRs #2170, #2186, and #2192, this issue is complete. Closing now.

closed 2025-04-06 by davidhutchens