logisim-evolution/logisim-evolution #2172: Glitchy overlapping UI

Wilki406, 2025-03-24:
I have encountered a bug where the UI overlaps and glitches out on me when i hover my cursor over UI elements as seen below

I have just reinstalled Logism Evolution 64bit for windows from this github and have the latest version of java installed.

[image]

maehne, 2025-03-24:
Thanks for the report! This is most likely a duplicate of issue #2076, which was [fixed](https://github.com/logisim-evolution/logisim-evolution/issues/2076#issuecomment-2431984207) with an update of the Flatlaf dependency. Could you please try to install the latest [Nightly build](https://github.com/logisim-evolution/logisim-evolution#nightly-builds) and report back whether your issue is indeed already fixed?

Wilki406, 2025-03-26:
> Thanks for the report! This is most likely a duplicate of issue [#2076](https://github.com/logisim-evolution/logisim-evolution/issues/2076), which was [...]

Yes that seems to have fixed the issue, Thank you!

closed 2025-03-26 by Wilki406