logisim-evolution/logisim-evolution #2175: Dragging a RAM component resets its contents.

davidhutchens, 2025-03-25:
Dragging a RAM component to a new location on the canvas resets the RAM contents in v3.9.0. I noticed when checking things while upgrading the circuit propagation components.

I also checked to see that it does not reset the contents in the original Logisim. Somewhere along the way, it stopped working as intended. I will see if I can figure out where that happened and what might be done to fix it.

maehne, 2025-03-25:
Thanks @davidhutchens!

parceval, 2025-08-28:
This is the same for all memory like component, also flip flops, register and shifters, they loose their content when dragged around in the canvas, because this state belongs to the simulation and not the component itself, opposite to the ROM component.

davidhutchens, 2025-08-28:
You are correct, @parceval. Yes, it is also related to the fact that the value is held in the simulation state.

However, they do retain their value in the original Logisim. There is even some code with comments suggesting it is there for this purpose, but it isn't triggered for some reason. So this is a feature that was lost at some point in Logisim-evolution prior to 3.9.0.

davidhutchens, 2025-08-28:
I just downloaded jars from our releases. The value is retained when dragging in LE 2.15.0. It is not retained in LE 3.3.1. LE 3.3.0 isn't running for me. So the feature was lost between 2.15.0 and 3.3.1. It will take time to narrow it further and figure out why.