logisim-evolution/logisim-evolution #2177: Timing diagram shows slower performance of the CSA

mk1nz, 2025-03-26:
I am studying and exploring different types of adders, and have found that the speed of the carry-select adder is lower than the ripple-carry adder. I'm not sure if this is a problem or if I'm doing something wrong. But could you please check this comparison of timing diagram for both types, since carry select adder is known to be the fastest.

[image]

[image]

BFH-ktt1, 2025-04-01:
Hi @mk1nz : There are two main problems here:

1) Logisim-evolution was never designed to incorporate "correct" gate delay's, the gate delay's incorporated are there to prevent the simulator to dead-lock. Hence the timing results should be taken with some care.

2) A CS-adder only makes sense when you can mitigate the delay's of the multiplexers with the delay in the different chains. In your example you only have a 1-bit "base" that selects the multiplexer result on two branches with 3 adders, this is by definition slower as a normal  CR-adder

Hope this helps

BFH-ktt1, 2025-04-01:
Closing this issue as it does not represent an enhancement nor a bug in logisim. Please re-open when you feel that there is a problem

closed 2025-04-01 by BFH-ktt1