logisim-evolution/logisim-evolution #2181: RegTab is causing simThread to lockup.

davidhutchens, 2025-03-30:
PR #2180 claims that RegTab is causing the simulation thread to lock up. I haven't verified that, but I can lock it up with the proposed fix in #2180.

In either case, we should examine the changes in the Holycross fork to update how RegTab works. It looks like there are some substantial changes to how it works there.

redweasel, 2025-04-03:
AFAICT the holycross version([kevinawalsh/logisim-evolution/.../RegTabContent.java](https://github.com/kevinawalsh/logisim-evolution/blob/fb2d5749e120b4e7314fa5b799e7598b436ab5a7/src/com/cburch/logisim/gui/generic/RegTabContent.java)) has implemented listeners for changes in circuits and projects and uses that to optimize the register tab. However, the misuse of the gui thread is causing the major slowdown.
I have looked at the holycross version and it appears to modifiy the JLabel from the simulation thread. Swing is not threadsafe, so that is UB. We should not copy that!

I have fixed the locking up in my PR branch by introducing a flag to stop spamming the gui thread. I'm also trying to port the other changes from the holycross version right now.
Can we reopen my PR, or should I create a new one?

One question from trying to port the holycross version: What is the "prodis_v1.3" library and why is it a special case in the register tab handling? (The holycross version has removed it)

davidhutchens, 2025-04-03:
@redweasel: I assigned this issue to me. I have been working on it. I agree that there is still a piece of interface in the simulation thread in the holycross version. I have been considering ways to remove it. Logically, the grabbing of the value for the drawing should only be happening as often as the drawing is being done (20 frames per second) and it should be done from the drawing thread.

Possible solutions: 1) Canvas could call RegTab to update values before initiating the drawing cycle. 2) RegTab could stop using JLabels and any other gui components and just write the data in a clean frame when asked to draw. It could look them up at that time. There may be better options I haven't thought of yet.

The string prodis does not occur anywhere in my src directory. I'm not sure where you are seeing that.

Please let me consider options and try to solve this. If you have specific suggestions, post them here. Since I have already ported the things from holycross, you are wasting time repeating that effort.

redweasel, 2025-04-03:
There is the prodis string:
https://github.com/logisim-evolution/logisim-evolution/blob/b6dbe9c4319ef71a53ce8e8dd3a66134458fadf5/src/main/java/com/cburch/logisim/gui/generic/RegTabContent.java#L66C5-L66C94

davidhutchens, 2025-04-03:
@redweasel: By the way, with my current build after taking things from holycross, a simple circuit with just a register with value fed through a add 1 and back to the register so it increments on each clock tick runs at over 700 kHz on my Mac M2 while showing the register through RegTab. It runs at 1.25 MHz (yes that is an M) when hiding the RegTab pane. A lone clock will run at 2.7 MHz.

davidhutchens, 2025-04-03:
The prodis library part went away, along with the project circuit list because the info is now being built from the circuitState tree rather than the project circuit list. The tree is the appropriate way to get the registers in the simulation since it accounts for circuits being used more than once in the design.

davidhutchens, 2025-04-04:
I have a fix that removes setting the Label text from the propagationCompleted method in RegTabContent. Canvas calls it to set the labels at redraw time. That speeds up the incrementing register mentioned above showing the register to almost as fast as it runs when not showing the register.

I still have to make the code meet our style standards and do some more testing. I hope to have a PR in a few days.

davidhutchens, 2025-04-06:
With the merger of PR #2192, this issue is complete. I will close it.

closed 2025-04-06 by davidhutchens