logisim-evolution/logisim-evolution #2182: Splay causes java.lang.StackOverflowError for big designs

BFH-ktt1, 2025-03-31:
With the new improvement in the simulation core, big designs cannot be simulated any more as they cause a stack overflow:
```
java.lang.StackOverflowError
        at com.cburch.logisim.util.SplayQueue.splay(SplayQueue.java:119)
```
@davidhutchens : Could you look into this point, I can send you some of our typical designs. I cannot put them here in the issue, as we use them as graded projects for our students. Could you tell me where I can put the example designs?

davidhutchens, 2025-03-31:
That is surprising. First, you should be able to fix the problem and get back running by changing the declaration in ../logisim/circuit/Propagator.java, line 130, to PriorityEventQueue and recompiling. That should go back to Java's PriorityQueue.

If the project will fit in an email, you can email it to me at david DOT hutchens AT millersviile DOT edu.

davidhutchens, 2025-04-01:
The Splay Tree implementation of the simulation event priority queue is not as balanced as it needs to be in some cases. The result is that some parts of the tree are far from the root. Since the implementation is recursive, it floods the stack and dies.

We should go back to Java's PriorityQueue implementation for our default. It is not the fastest in many cases, but it handles extreme cases better. I'll include that in my next PR.

I am considering allowing the user to choose the implementation. If I do that, should that choice be saved in individual .circ files, or should it be a user level preference setting? Or should I not bother doing it at all?

BFH-ktt1, 2025-04-01:
Hi @davidhutchens : I would do the choice as user-level preference. I think it is nice that the user can select it. For Splay we could for the moment add the experimental tag. Thanks for looking into this issue!

maehne, 2025-04-01:
@davidhutchens: Did you have a look at commits https://github.com/kevinawalsh/logisim-evolution/commit/034c7cbc9ca12c64d7c1daa910aba7aca752fac8 and https://github.com/kevinawalsh/logisim-evolution/commit/bee8d6d620cf54e22817e04aba6949ac9cc53080? In them, @kevinawalsh mentioned the issue regarding the splay queue. Also, the commits following these two seem to be relevant to @kevinawalsh's effort to improve simulation performance.

davidhutchens, 2025-04-01:
@maehne: Yes, I noticed those. They were also included in the commits in PR #2170. Those choices were what I was asking about above: if I should make them available in the user interface. In my tests, the splay queue is faster than the others. And the linked queue was the slowest. Kevin had the linked queue as the fastest and the priorityqueue slowest. There appears to be differences depending on the projects. I'm not sure why.

maehne, 2025-04-02:
Thanks for your feedback @davidhutchens!

davidhutchens, 2025-05-07:
I have spent some time looking into why the queues perform differently based on projects. Here is my analysis.

A simulation event has two keys used for ordering them in the priority queue. The major key is the timeKey. It specifies the simulation time step in which the event should be processed. It is calculated as the component delay plus the current time. The number of distinct times is relatively small and limited by the maximum component delay. For my big computer simulation there are fewer than 50 distinct times in the queue at any time and quite often much fewer than that. For the large projects that @BFH-ktt1 sent me, noted above, there are generally fewer than 12 at any time.

Each timeKey has simulation events associated with it. Sometimes it is just one or a few. Sometimes it is a very large number. Most of the events target the smallest timeKey (current time + 1) since that is the delay for simple gates, pins, wires, etc. For example, in my big computer simulation more than 80% of the events are placed in the least timeKey at the time of generation. For the two projects @BFH-ktt1 sent me, the numbers are over 90%.

The simulation events receive a serialNumber when they are generated and added to the queue. These numbers are generated sequentially, adding one to the number for the next event. The serialNumber is the second key used for ordering the events in the priority queue.

Priority queue implementations typically have roughly log(n) time for each insertion or removal where n is the number of items in the queue. Since there are a lot of simulation events, n can be rather large and the queue can use a significant amount of time in the simulation cycle. The Linked Queue and Splay Tree Queue are attempts at doing better at a potential downside of poor worst-case performance.

So the main factors that determine a particular queues efficiency is how it deals with the two keys and how they interact in the project.

Thinking about the situation, it occurred to me that one could implement a two level priority queue. The first level uses the timeKey in a sorted map (either a simple linked list or a Java TreeMap) containing TimeNodes that each has a simple singly linked queue of events. Within each timeKey, the arrival of all the events is in order of the serialNumber and thus all of them can be added to the back of the queue in constant time (once you locate the TimeNode). Using this approach, the log(n) component is at most the lookup of the TimeNode of which there are very few.

Furthermore, since the significant majority of the adds go in the smallest timeKey, that TimeNode can be cached outside the map and all those adds can happen in small constant time. That also includes all peeks and removals since they all happen at the smallest timeKey.

I implemented this QueueOfQueues approach with a choice of a linked-list or TreeMap lookup for the TimeNodes. Both are faster than the other three implementations on all but the simplest of projects (a single clock, or a very small FSM). They are only 3% slower than the Splay Tree on a single clock and 1%-3% slower than the Linked List on a small FSM. Note that those differences are barely measurable but make some sense. A single clock has one event and the splay tree adds one event with a single reference assignment.

Because the number of distinct timeKey values remains small for most projects, the linked-list map actually runs 0%-4% faster than the TreeMap on the real projects I tested on. I did create a project designed to scatter the events over many TimeNodes and not concentrate them at the earliest ones. The linked-list map dropped back to the other queues. But the TreeMap handled it well.

So I am finishing work on allowing the user to select among the 5 priority queue implementations. One of the QueueOfQueues choices is likely to be most efficient. A PR will follow after I am happy with it.

davidhutchens, 2025-05-07:
I didn't actually say how much faster the QueueOfQueues implementation is over the other three queues. Using my computer projects and @BFH-ktt1's examples for the reference, the two QueueOfQueues are 10%-30% faster than the Splay Tree, 20%-60% faster than the Java PriorityQueue, and 20%-230% faster than the LinkedQueue.

Note that I did run @BFH-ktt1's examples using the Splay Tree. I had to increase the runtime stack size from the default 2MB to 16MB to get them to run. Even with that deep recursion, it was still faster on those projects than either the Java PriorityQueue or the LinkedQueue! But the QueueOfQueues wins.

Also note that the QueueOfQueues will not stress the runtime stack. The linked lists are iterative. If used, the Java TreeMap is a red-black tree, which is guaranteed to be balanced, and the TreeMap has a relatively small number of nodes anyway.

BFH-ktt1, 2025-05-27:
Closing this issue due to merging of #2232