logisim-evolution/logisim-evolution #2187: defect: HDL clk signal

primechaser, 2025-04-02:
Hi Team,

the latest nightly build (2025-04-02) introduces a potential defect regarding the use of HDL entities using the clk signal.

The attached test circ shows 2 simple HDL entities

a) HDL_issue
b) HDL_OK

when looking at the final code, the following is getting generated

....
   HDL_issue : hdl_wrong
      PORT MAP ( CLK  => s_logisimBus0(7 DOWNTO 0),
                 DIn  => s_logisimNet1,
                 Dout => s_logisimBus2(7 DOWNTO 0) );

   HDL_OK : hdl_right
      PORT MAP ( CLK  => s_logisimNet1,
                 DIn  => s_logisimBus0(7 DOWNTO 0),
                 Dout => s_logisimBus3(7 DOWNTO 0) );

this implies for any HDL entity using a clk signal it needs to be defined at the very bottom of the entity definition, as otherwise mappings will be wrong.

Thanks for looking into this

[test01.zip](https://github.com/user-attachments/files/19563806/test01.zip)

maehne, 2025-04-02:
This error occurs during netlisting for FPGA synthesis. 

@BFH-ktt1: Could you please have a look? Attached, you can find an updated test case, which includes already the mapping for our GECKO4-EDUCATION.

[test01.zip](https://github.com/user-attachments/files/19564893/test01.zip)

BFH-ktt1, 2025-04-04:
This should be fixed by my PR  #2188

BFH-ktt1, 2025-04-06:
Closing as it should be fixed with my merged PR #2188. Please reopen if you feel that the problem is not resolved.

closed 2025-04-06 by BFH-ktt1