logisim-evolution/logisim-evolution #2194: Canvas does not return from dark mode when going back to light mode

rshouker, 2025-04-07:
I tried the dark mode, things didn't look right (that's another issue.) I return to light mode, restarted Logisim Evolution, the canvas was still black. I uninstalled LE, restarted my computer (windows 11,) reinstalled LE, that didn't help.
Also, I set the colors in preferences->window, and restarted LE; didn't do the trick.
I repeated the process, this time eliminating any reference to logisim in the **registry,** before restarting my computer and reinstalling LE; this worked.

maehne, 2025-04-09:
I am not able to reproduce the described behaviour on Linux. After changing the Look and feel in Window -> Preferences -> Window, I needed to simply restart Logisim-evolution to either have a dark or light look and feel. 

Did you test whether the [current nightly releases](https://github.com/logisim-evolution/logisim-evolution#nightly-builds) improve the situation?

However, I agree that the user interface is currently not properly coloured in all cases to make a dark theme a viable option. If I remember correctly, @MarcinOrlowski worked primarily in this area. Maybe, he can look into this issue. PRs to enhance the user experience when using a dark theme are welcome!

hewzhew, 2026-08-29:
Retested on Windows using current `main` (`c15ac4bf1`). Switching from `Flat Dark` to `Flat Light` immediately restored both the UI and canvas to the light palette. After closing and relaunching Logisim-evolution, both remained light.

This appears to have been addressed by #2655. Older installations with already inconsistent stored colour preferences may still need one explicit theme switch (or a colour reset), but the dark-to-light path reported here no longer reproduces in the current version. Would it be reasonable to close this issue as resolved?

davidhutchens, 2026-08-29:
Closing as completed by #2655. If problems remain, please reopen.

closed 2026-08-29 by davidhutchens