logisim-evolution/logisim-evolution #2196: Clock pin rendering

kjcolley7, 2025-04-08:
Before #2158:

[image]

After #2158:

[image]

Note that clock pins are no longer displayed last (which I guess is the main intent of that PR), but also clock pins now always display the label even when there's only one clock pin. Is that how we'd like it to work?

Another option to preserve backwards compatibility while supporting the nicer (IMO) rendering with clock pins at the end and labels only displayed if there's multiple clock pins would be to add a new persisted pin attribute for "isClock". Existing logisim files without this attribute stored would have it default to false, and newly created pins would have it default to true if the pin's name matches the clock pattern, or false otherwise. That way, existing documents will load all pins with isClock=false, meaning they'll render as normal pins. But any newly created circuits with a pin with "clk" or "clock" in the name will automatically render as a clock pin sorted to the end.

Thoughts?

kjcolley7, 2025-04-08:
Also if the HDL bug was because of the sorting, then we could perhaps keep the original sorting behavior but instead add a new sorting method that is only used by the rendering code. So the HDL code will use the original sort order and the rendering code will use the new one that puts clock pins last.

maehne, 2025-04-08:
@BFH-ktt1: Could you have a look at this issue and comment, as you accepted the original PR and provided a fix for the HDL generation?

Personally, I prefer that the port name remains visible and that the ports are sorted according to their vertical position in the circuit sheet instead of their role associated role as a clock. Hiding the port name if there's a single clock input is acceptable for me, too. One important aspects is also backwards compatibility: Are block symbols automatically modified, if one opens older designs created before your enhancement PR? If no, pushing down clock ports in the symbol would be fine for me. However, it is important that the HDL gets generated correctly in all cases -- also with VHDL sub-circuits.

By the way, what about falling edge clock inputs? Did you consider recognizing them, too, on based on some naming convention, e.g. `_n` or `_b` suffix, to indicate graphically the edge sensitivity?