logisim-evolution/logisim-evolution #2198: Feature request: Either allow dip switch of a single switch, or have a separate single switch

rshouker, 2025-04-08:
Switches are very useful, especially since there is no, afaik, way to lock a button.
Dip switches have a minimum number of switches set to two; I need a single switch either as a dip switch or as a toggle botton (can be its own entity, or as an option for the button.) Yes, I can connect only a single switch in a dip switch, but as a teaching tool it would be nice if i had a nicer choice.

maehne, 2025-04-09:
For this purpose, I usually use simply input ports, which can be easily set to the desired value. They also allow adjusting the representation in case of multiple bits.

BFH-ktt1, 2025-05-27:
I'm closing this issue, as no follow up. Please reopen it if you feel that the issue is not solved.

closed 2025-05-27 by BFH-ktt1

sinesc, 2025-07-25:
How is this issue closed and why does it need any follow up? There is no single switch. Adding an input puts the input on the component. Why not just allow single dip switches?

PR #2306 (2025-08-10): Allow DIP switch minimum width of 1
This change modifies the DIP switch component to allow a minimum width of 1 instead of 2.

Summary of changes:
- Updated MIN_SWITCH in src/main/java/com/cburch/logisim/std/io/DipSwitch.java from 2 to 1.
- Added an inline comment explaining the change.
- Verified rendering, bounds, port positioning, and validation logic all work for width=1.
- Successfully built and ran the application to confirm no regressions.

Enables creation of single DIP switches for simple on/off controls and single-bit teaching examples.

Fixes #2198