logisim-evolution/logisim-evolution #2199: Feature request: SOC exposer

AlphaTechNinja, 2025-04-10:
I have a new SOC chip idea that would be wonderful. the SOC exposer the way this would work is kind of like the Parallel Input/Output expander but instead of outputting the written value it takes the address and outputs it on the bottom. Depending on the mode then either reads from the top port or writes after a certain delay in case it takes time for the circuits to properly update their pins. The reason I ask for this chip as it would allow for some more complex designs one example I can think of is wiring one of the RAM chips to exposer (or EEPROM) using it as if it were connected to the SOC bus it could also allow connecting a custom communication system between SOC processors. Please let me know if this idea will be added or possibly be added in the future

maehne, 2025-04-26:
Your enhancement request requires discussion. If I understand it correctly, you basically want to expose a cycle-accurate bus interface to which Logisim sub-circuits could be attached as bus transaction targets. This will potentially have a big impact on performance, as currently bus transactions are basically always taking a single clock cycle, which won't be guaranteed with a typical gate-level implementation of a bus peripheral.

@BFH-ktt1: What are you thoughts?

AlphaTechNinja, 2025-04-26:
Sounds about right. I only have one idea at the moment to try to get it to work but I'm not sure how well it integrates into the existing simulation maybe just have it update the circuits connected to the address pins but I'm not familiar with the internal workings of this simulator so I'm not sure how bad performance will dip. Going this route will definitely lead to some simulation impacts if this element is connected to large circuits.
(I'll probably read through the SOC code to see if there is any possible way to limit the impact)

Maybe others have some better ideas of how to implement this if it were to exists

BFH-ktt1, 2025-05-30:
For me this does not make any sense, as the engine behind the SOC-components are not related to hardware, but to software, meaning the bus is not modeled as in gates, etc. it is just a very simple insert-activate engine. This is also why it is not supported in HDL; read downloadable on hardware. It is an "extra simulator" I added, as it was easy in java, for a course we are giving. Changing this part in something that could be eventually be "downloadable" on hardware, requires a big effort. Something I never intended for it. So I'm closing this issue, as I think it goes far out of the scope of the intention of this addition. Any PR's in a new direction are welcomed.