logisim-evolution/logisim-evolution #2207: Gradle V 8.13: Deprecation warning "Invocation of Task.project at execution time has been deprecated.

xvvasan, 2025-04-18:
We have tried deploying using mlReloadModules with --warning-mode all option using gradle Version 8.13 and getting the following deprecation warning:

// begin-report-data
{"diagnostics":[{"locations":[{"taskPath":":mlDeleteModuleTimestampsFile"}],"problem":[{"text":"Invocation of Task.project at execution time has been deprecated."}],"severity":"WARNING","problemDetails":[{"text":"This will fail with an error in Gradle 10.0."}],"contextualLabel":"Invocation of Task.project at execution time has been deprecated.","documentationLink":"https://docs.gradle.org/8.13/userguide/upgrading_version_7.html#task_project","problemId":[{"name":"deprecation","displayName":"Deprecation"},{"name":"invocation-of-task-project-at-execution-time","displayName":"Invocation of Task.project at execution time has been deprecated."}],"solutions":[[{"text":"This API is incompatible with the configuration cache, which will become the only mode supported by Gradle in a future release."}]]}],"problemsReport":{"totalProblemCount":1,"buildName":"dofri-marklogic","requestedTasks":"mlReloadModules","documentationLink":"https://docs.gradle.org/8.13/userguide/reporting_problems.html","documentationLinkCaption":"Problem report","summaries":[]}}
// end-report-data

Earlier, we tried with gradle Version 8.10 and that was working fine. some one can help to resolve this, Thank you.

davidhutchens, 2025-04-18:
Did you use the current main branch? We have upgraded it to Gradle 8.12.1 and I think we fixed those issues at that time. Gradle 8.13 worked also when I tried it on the main branch. I am waiting for Gradle 8.14 to upgrade it again since 8.14 should support Java 24. Java 24 does issue warnings from Gradle about Gradle at this time.

If this happens on the main branch, can you provide more detail as to your environment?

davidhutchens, 2025-04-19:
@xvvasan: Did you post this issue to the correct place? Logisim-Evolution does not include the mlReloadModules task in its gradle configuration.

davidhutchens, 2025-04-26:
I will close this as resolved with our upgrade to Gradle 8.14 in PR #2213.

If the issue remains and you can give detailed instructions for us to reproduce it, please reopen this issue.

closed 2025-04-26 by davidhutchens